PaulStoffregen/Time #9: Retry for Sync

wayfarerbasta, 2015-02-06:
Hi Paul,
thank you for the great Library. I'm using it with a Texas Instruments CC3200 Launchpad and Energia IDE.
I have an issue with the setSyncProvider function. I'm using an NTP SyncProvider function. In the rare cases, when the NTP server fails to deliver the correct Unix Timestamp, the function returns the value 0.
In this case I would expect that time will try to resync at least after the 300 s, the default value for SyncInterval. Unfortunatly is does not work. Further I would prefer to set a retry Interval in Cases of a sync failure. Do you have any advice for me

Tanks 

wayfarerbasta

PaulStoffregen, 2015-02-06:
Are you sure it really stops trying to sync if the return value is zero?

wayfarerbasta, 2015-02-06:
Hi Paul,
I double checked it and, I don't know why, but now it works. I just added an additional variable to your cpp to maintain the Sync interval and retry Sync interval seperately.
Thank you.

closed 2015-02-06 by wayfarerbasta