PaulStoffregen/Time #13: Why a while-loop?

karl-petter, 2015-04-08:
Considering using this library to represent time I receive from a GSM module using AT commands. To see if it would fit my needs I took a look at the code and found one thing that looked confusing to me. And there were no comment to explain why.

The now() method starts with this while-loop:
```
  while (millis() - prevMillis >= 1000){      
    sysTime++;
    prevMillis += 1000;	
#ifdef TIME_DRIFT_INFO
    sysUnsyncedTime++; // this can be compared to the synced time to measure long term drift     
#endif
  }
```
It seems to calculate number of seconds that has passed since last call to now() and add to sysTime, correct? If so, why not just write it this way?
```
sysTime += (millis()-prevMillis)/1000;
```
I find that much easier to understand what is going on.

michaelmargolis, 2015-04-08:
`sysTime += (millis()-prevMillis)/1000;`
 
That would  require additional code to check if the sysTime has rolled over and  if so,  update prevMillis with the  current millis value. millis would either need to read again or temporarily stored in another variable.  That considered, I prefered the form as written.

Michael Margolis

PaulStoffregen, 2015-04-08:
I agree, the integer division approach would almost certainly introduce a bug into the library.  It may be conceptually easier tor read, but what good is that if it fails in some cases?

karl-petter, 2015-04-08:
Just wanted to point out some comments in that section would help the readability of the code. And why it was written this way.

karl-petter, 2015-04-09:
Created a pull request with two comments added. https://github.com/PaulStoffregen/Time/pull/14

I do not know your view on comments in code but personally I like to have them added when the code is not obvious directly and I think this is such a case since one needs to know in detail how subtraction of unsigned ints work to realise what is going on. But just reject it if you feel it clutters up.

quantenschaum, 2015-11-05:
The while loop runs unnecessarily long if now() is not called often.  Why does

    sysTime += (millis()-prevMillis)/1000;

require a check for `sysTime` roll over? The while loop increments `sysTime` n times, which is the same as adding the number of seconds since last call of `now()`. It rolls over when it rolls overs, same in either case. I would replace the while loop by

    uint32_t n = (millis() - prevMillis) / 1000;
    sysTime += n;
    prevMillis += n * 1000;

`n` will always be positive and is exactly the number of times the while loop will run. The integer division result is floored, fractional parts are simply discarded. This is what we want here. I do not get the point where the division would introduce an error or where the rollover must be checked for. It will still work when `prevMillis` rolled over and is not a multiple of 1000 anymore, because then `millis()` rolled over, too. 

One could also ask, if

    millis() - prevMillis >= 1000

is always the same as

    (millis() - prevMillis) / 1000 > 0

If it is, and I think it is, then one can replace the while loop by the code above.

On the other hand the returned time would be wrong, if `sysTime` rolled over, but the while loop does not prevent this from happening. Or am I getting something wrong here?

quantenschaum, 2015-11-05:
btw unix time (unsigned 32 bit int) will roll over Sun, 07 Feb 2106 06:28:15 GMT

So `sysTime` rolling over might become an issue, but no yet.

arkhipenko, 2015-11-14:
Absolutely agree with quantenschaum:

    uint32_t n = (millis() - prevMillis) / 1000;
    sysTime += n;
    prevMillis += n * 1000;

is the way to go. Unsigned long arithmetic will take care of the rollover automatically.

michaelmargolis, 2015-11-14:
The loop construct was used in anticipation of a yield method being added to allow the addition of cooperative service routines (for example, see the delay function in the latest  Arduino core). The loop  would ensure that code hooked into the now() function would get  a call for each elapsed second since the previous call.

arkhipenko, could you clarify your use case where 3600 cycles are needed. If the sketch is restarting then prevMillis will be set to the current millis so only one cycle will be needed. If the sketch is running but idle waiting for an hour to elapse,  then is there a case not do do something like the following in the loop code?
```
   // code to do something here... 
   ...
    time_t prevTime = now();
    while( now() < prevTime +3600)
           ; // nothing to do
     // here after an hour, repeat loop
```
Please do clarify if I have missed something here

Michael

arkhipenko, 2015-11-14:
Hi Michael,

The use case is: I use TaskScheduler for cooperative multitasking. The task is scheduled to run every hour, and the first thing it does is check what time it is, for which it calls now().
Since there was no calls to now since last hour, counting up to current millis() will take 3600 runs through your loop (one second at a time).  Since I already use task scheduling framework, I do not need to use your suggested code. In fact, the main scheduling loop of TaskScheduler has been optimized such that it tries to waste as few cycles as possible, so I am not using any external libraries and/or time sources - just millis(). 

However, this was not the point.
The point was that your loop does not seem to be necessary at all.  Quantenschaum's code will achieve the same result in one pass with predictable execution time. There is no issue with rollover due to unsigned arithmetic (unless now is not called for 47 days and it rolls over max number for unsigned long integers altogether). 

Let's see the example with regular unsigned int:

    prevMillis = 65000U;
    // unsigned int rolls over at 65535
    // time elapsed 3600 millis, so millis() rolls over to 3065
    // 3065 - 65000 = -61935 signed = 3600 unsigned 16 bit
    uint32_t n = (3065 - 65000) / 1000 = (3600)/1000 = 3
    sysTime += 3;
    prevMillis += 3000;

Same result. One pass. Current code would need 3 loop cycles to arrive at the same result.
What am I missing?

michaelmargolis, 2015-11-14:
I think you missed my point that some people might wish to hook a cooperative multitasker into the time libraries timekeeping,  which is the reason for the loop. That said, the support for a hook is not in the code and if people such as yourself that are building multitaskers don’t feel the need to integrate closely with the time library then perhaps it is not really needed.

To better  understand your use case, I would like to see  your sketch.

After a quick look at TaskScheduler, I wonder your proposed change would work in all cases.  
How  would a sketch  maintain time if _TASK_SLEEP_ON_IDLE_RUN is defined?  

Perhaps we should continue this discussion elsewhere and come back here after the issue is resolved. Do you have a thread in the Arduino Forum where we can discuss TaskScheduler in more detail.

quantenschaum, 2015-11-14:
I do not get the point why the loop is better than my approach. If you implement as I proposed, the result will be exactly the same, but it will execute in constant time.

@PaulStoffregen Why would integer division introduce a bug?

@michaelmargolis I do not understand your task scheduler thing. What does this have to do with the to-loop-or-no-to-loop question?

quantenschaum, 2015-11-14:
@michaelmargolis What hook are you talking about? There is no hook in this library and I am pretty sure, that could achieve what you want without **the loop**.

michaelmargolis, 2015-11-14:
quantenschaum, as I posted above, the loop code allows for the addition of a hook so a multitasker could get a callback once for every second since the time library starts running. I also posted that this feature has not yet been implemented and I will revisit if there is an actual need for this.

 I do want to understand arkhipenko's application as this may relate to the use cases I had envisaged for the  multitasking hook.  That specific discussion probably does not belong here but I will return to this discussion after arkhipenko and I explore ways to use his multitasker with the time library.

arkhipenko, 2015-11-14:
Ah! A lively discussion... love it!

About multitasking. As I understand it, the purpose of the Time::now() method is to return current system time (optionally syncing it with the clock source if the time has come to do so).  

Do we agree on that?

If so, the only contribution to cooperative multitasking for such method would be to return as fast as possible.
The integer division achieves just that. The loop introduces blocking of a variable duration, which is actually against to the spirit of cooperative multitasking.  Introducing a multitasking "hook" into the method (which just needs to return a value as soon as possible)  has questionable benefit in my mind: if a task needs current time, having an option to yield to it again does not make sense for that task - it is waiting for the current time...  In fact, if you start yielding inside now() you will make that method return even later! At which point we will have to rename it from now() to something like someTimeLater()... :0

Perhaps there is another purpose behind now() which I did not understand?

PS: About TaskScheduler - happy to discuss in on the forum. I don't have a dedicated thread for it, but we can create one.  
Specifically about TASK_SLEEP_ON_IDLE_RUN: this compile option puts avr to IDLE state, which keeps millis() running, waking the device every once 1 millisecond, So as someone put it quite nicely, it is not a sleep, but a series of "very quick naps"
Check out TaskScheduler example N6, which illustrates how IDLE affects execution under TaskScheduler.

michaelmargolis, 2015-11-14:
The hook would be valuable when one wants  multitasking tightly integrated into time of day (as provided by the time library).   See TimeAlarms for an example: https://www.pjrc.com/teensy/td_libs_TimeAlarms.html

That library uses explicit calls instead of a hook , but the principle is that scheduled events are always  processed in the correct sequence for the correct number of times,  even when the system 'wakes up' after these  events were supposed to be scheduled.  Your library has a different approach but that does not mean that any other way is wrong.

Please do open a thread so we can discuss in detail.

quantenschaum, 2015-11-14:
Sorry @michaelmargolis, where is the hook?

now() reads
```C
time_t now() {
	// calculate number of seconds passed since last call to now()
  while (millis() - prevMillis >= 1000) {
		// millis() and prevMillis are both unsigned ints thus the subtraction will always be the absolute value of the difference
    sysTime++;
    prevMillis += 1000;	
#ifdef TIME_DRIFT_INFO
    sysUnsyncedTime++; // this can be compared to the synced time to measure long term drift     
#endif
  }
  if (nextSyncTime <= sysTime) {
    if (getTimePtr != 0) {
      time_t t = getTimePtr();
      if (t != 0) {
        setTime(t);
      } else {
        nextSyncTime = sysTime + syncInterval;
        Status = (Status == timeNotSet) ?  timeNotSet : timeNeedsSync;
      }
    }
  }  
  return (time_t)sysTime;
}
```

I do not see any hook here and thus the loop does not make any sense to me.

arkhipenko, 2015-11-14:
@michaelmargolis 
OK. I see what you are trying to achieve, but I still believe now() is the wrong place to achieve it.
now() should return current system time. period. and do it fast. 
A ticker functions which invokes tasks at specific times and in specific order should be a separate method.

arkhipenko, 2015-11-14:
@michaelmargolis 
Another problem with current approach is that you will force your multitasking into 1 second ticker resolution, which could be too long for certain functionality. (For instance I read gyroscope readings for one of my robots every 10 milliseconds).

michaelmargolis, 2015-11-14:
>  where is the hook?
@quantenschaum, please take the time to read my posts, I answered this your question in two previous replies.

michaelmargolis, 2015-11-14:
@arkhipenko,  The library was created for applications where a one second precision is suitable.
Your application is the first I have come across where this may be a problem.  I am keen to understand more about your application and will have a look at your code over the coming days. But I prefer not to extend the discussion on that here as it may not be relevant for the majority of users of the Time library.
We can come back here when we have a resolution.

quantenschaum, 2015-11-14:
@michaelmargolis I did read them. In your previsous posts you talk about something hypothetic, that "the hook would be valuable" and that "loop code allows for the addition of a hook". I was look for something that actually exists, but "the hook" you are talking about does not exist.

I agree with @arkhipenko and when I look at what now() currently does, the loop does not make any sense. The time library should keep track of the time, that's it, point. Anything that has to with task scheduling should be handled in the task scheduling library, which in turn may make use of the time library, but it must not creep into the time library.

Given the current behavior of now(), I could modify the implementation (remove the loop) and it will behave the same, seen from the outside, but more efficient.

Another problem I see and discovered in my projects was, that now() calls the time provider function, which may take long to return and also may use resources, that are not available at every place in the program where now() is called. Or the time provider may interfere with other things if invoked from certain invocations of now(). So it would be desirable to be able to turn off automatic syncing within now() and do the syncing at well defined place. Invoking the time provider within now() also violates the keep-it-simple paradigm. now() should return current time not perform any fancy stuff with funny side effects. There should be updateTime() method (like ethernet maintain()) that is called at a well defined place in regular intervals and syncs the time using the time provider if necessary.

Same for setSyncProvider() and setSyncInterval(), they do more than just setting the values. Unexpected and potentially unwanted side effect.

blackketter, 2015-11-14:
I'd vote for removing the loop in the name of making now() fast and predictable. 

I also support  @arkhipenko's suggestion for adding support for sub-second resolution via separate function call.  This would be very useful for me. (For example, it simplifies the implementation of a sweeping second hand on a clock...)

arkhipenko, 2015-11-15:
@blackketter
Check out TaskScheduler.  :)
Subsecond precision and more ...

michaelmargolis, 2015-11-15:
For those looking to save a few microseconds,  it is interesting to do a comparison of the performance of the proposed code.  The examples below simulate an interval of five seconds between calls to now,  although in my experience, most sketches call now()  at least once per second.  

Here is the current code with micros() timers added for benchmarking
```
  static uint32_t sysTime  = 0;

  uint32_t prevMillis = millis() - 5 * 1000L;  
  uint32_t startT = micros();  

  while (millis() - prevMillis >= 1000) {
    sysTime++;
    prevMillis += 1000; 
  }

  uint32_t endT = micros();
```
here is the proposed ‘faster’ code:
```
  uint32_t prevMillis = millis() - 5 * 1000L;  
  uint32_t startT = micros();

  uint32_t n = (millis() - prevMillis) / 1000;
  sysTime += n;
  prevMillis += n * 1000;

  uint32_t endT = micros();
```
Because 32 bit integer division is slow on the 8 bit Arduino boards, the current Time code is almost twice as fast in this example.  Above ten seconds between calls, the current code is slower, but for most users, I think that will not be the case.

blackketter, 2015-11-15:
TaskScheduler is nice, thanks!  

But it doesn’t provide sub-second wall clock time, does it?

michaelmargolis, 2015-11-15:
@blackketter,  the following isn't tested but here is some code if you want an indication of the number of milliseconds that have passed since the turn of the last second.
This gets added to Time.cpp
```
int millisecond() 
{
     return ( min(millis() - prevMillis,999)); // truncate  to prevent time running into the next second
}
```
add the declaration to time.h
```
int     millisecond();
```
usage in the sketch:
```
     Serial.print(second());
     Serial.print(".");
     Serial.println(millisecond());
```

arkhipenko, 2015-11-15:
@michaelmargolis:
There is a problem with your new  int millisecond() solution: it is deceptive.
And I think it underscores our discussion around the purpose of the library.

Deception: 
1. If millis() outruns prevMillis by more than a second, you just return 999 and no indication whether it is true 999 millis since the turn of the second, or it has overrun a second. (it maybe 233 ms into the next second, but will return 999)
2. Is prevMillis really syncronized to the turn of a second? I don't see where that is done in the code. You start counting when external source returns a value, not when the second turns.

This is how @blackketter 's problem should be solved:
1. Using a *scheduling library* a task should be scheduled to tick as many times per second as necessary (let's say 4 times per second for the sake of argument, or once every 250 ms).
2. Using *Time library* the exact time of the first invocation of the task from step 1 should be aligned with the turn of a second. 

For instance, using TaskScheduler, a task with 250 ms interval could be scheduled to run indefinitely.
In the OnCreate method of that task a synchronization with a turn of a second could be achieved by quickly querying now() and waiting for the seconds to turn.
Once the turn of second is detected, the task can use runCounter value to determine which of the quarter second run it is: 1: 0 ms, 2: 250 ms, 3: 500 ms, and so on.
Periodically task should resync with a turn of a second using Time library.

The main point is: two libraries are needed: one to tick, and one to provide the time. Mixing the two together is not a good idea.

michaelmargolis, 2015-11-15:
I am in complete agreement that the time library should not provide the functionality of a task scheduler. Nor will it provide sub second accuracy for a number of reasons. The one line routine I suggested is simply a way that a cosmetic sweep hand could be driven in somewhat consistent fashion.

blackketter, 2015-11-15:
I also agree that the Time library should not provide scheduling, but I do think that being able to obtain wall-clock time at better than one second resolution is a useful feature for the Time library.   

I'm not sure @michaelmargolis' solution is ideal, it should be possible to determine the an accurate sub-second time without having to test for overflow.

What I did for my own application (to get a sweeping second hand on a display) was to track when now() rolled over and note the current value of millis() and use that later on to calculate a millisecond offset.  A hack, really.

michaelmargolis, 2015-11-16:
With software timing on Avr boards, overruns into the next second will always be possible even if interrupts are used (for example if the second rolls over while some other hardware was servicing an interrupt) so a check is prudent.  
I would like to add direct support for real time clocks into a future version of the library which would  provide significant improvements in accuracy when RTC hardware is available.