PaulStoffregen/Time #16: Abbreviation of Thursday

karl-petter, 2015-05-14:
The abbreviation of Thursday in "ErrSunMonTueWedThrFriSat" on line 66 in DateStrings.cpp is Thr. Is this on purpose to be compatible with some specification? It is not compatible with the abbreviations used in RFC 2616 where Thu is used for Thursday. Caused me in interesting bugg which resulted in that my HMAC authentication worked on all days except Thursdays :)

grayl, 2015-05-20:
I agree with karl-petter.  'Thu' is the standard in the Canadian Forces, and as far as I know, all other English speaking forces in the world.

PaulStoffregen, 2015-05-20:
@michaelmargolis Any thoughts on changing this?  Seems like a pretty straightforward improvement.

michaelmargolis, 2015-05-20:
I would be happy if you can make the change. I am in the middle of a house purchase and move so things will be frantic here until I get settled
Michael
Date: Wed, 20 May 2015 14:25:04 -0700

PaulStoffregen, 2015-05-20:
fixed
https://github.com/PaulStoffregen/Time/commit/a7cc3261f91af37ca3d8c8e192fa4030445178e7

closed 2015-05-20 by PaulStoffregen