PaulStoffregen/Time #19: The 2038 problem...

vicatcu, 2015-06-05:
http://en.wikipedia.org/wiki/Year_2038_problem

Is it a problem for the Time library? Because it uses time_t as a core concept, which I believe is unixtime?

PaulStoffregen, 2015-06-05:
Yes, it will be a problem in 23 years

vicatcu, 2015-06-05:
OK feel free to close the issue, I was just wondering idly :o)

nobk, 2016-05-03:
a 3rd part time lib should avoid 2038 problem, otherwise why should i use this lib instead of standard C <time.h> lib? Is it possible to define a 'long long time_t' in your lib?

33r3, 2018-02-03:
I’ve only skimmed the code, but it looks like the library uses an unsigned 32-bit variable for the time instead of a signed 32. Shouldn’t that buy another several decades assuming whatever syncs the clock does so with that knowledge?

doanerock, 2018-03-12:
any update on the 2038 problem?

PaulStoffregen, 2018-03-12:
No, no update, but I can confirm for you this is a low priority right now.  You can interpret "low priority" to mean nothing will be done in 2018.

woutput, 2019-01-02:
I'd like to see this (confirmed by @PaulStoffregen) issue being solved. I expect that quite some people (including me) use this code for embedded systems. These systems may still be working and/or be difficult to update before 2038.
Also see [this](http://www.lieberbiber.de/2017/03/14/a-look-at-the-year-20362038-problems-and-time-proofness-in-various-systems/) and [this](https://sourceware.org/glibc/wiki/Y2038ProofnessDesign).
> nothing will be done in 2018

Maybe 2019 is a nice year to fix it :)
Let me know if I can help (for example by reviewing code and/or testing on ESP8266/ESP32/ATTiny85)

PaulStoffregen, 2019-01-02:
> Maybe 2019 is a nice year to fix it :)

Unlikely, working on a big hardware project.  At least for first half of 2019, all libs I maintain are more or less on a feature freeze.

woutput, 2019-01-02:
Can we then at least have very clear warnings (Readme.md, comment in source code, ...) that the code has this issue?

PaulStoffregen, 2019-01-02:
How I can give you the impression that I consider this a very low priority in 2019?

raffaeler, 2020-11-06:
According to the Wikipedia article:
> changing time_t to an unsigned 32-bit integer, which would extend the range to 2106 (specifically, 06:28:15 UTC on Sunday, 7 February 2106), would adversely affect programs that store, retrieve, or manipulate dates prior to 1970 ...

Currently the .h file already defines it as unsigned when it is not already found:
```
#if !defined(__time_t_defined) // avoid conflict with newlib or other posix libc
typedef unsigned long time_t;
#endif
``` 

Shouldn't be enough always defining time_t as an unsigned long to workaround the issue?

karelv, 2023-11-28:
Should be good until 2106...
Note: It is not only `time_t`, but also the variable `sysTime`, which of the type `uint32_t`.