PaulStoffregen/Time #21: Arduino DUE compatibility

lenny1972, 2015-06-23:
This library worked great on my Arduino Mega2560 but when I switched to Arduino DUE it doesn't work anymore.. Is it planned a new release which support also this board?

PaulStoffregen, 2015-06-23:
We tested this on Due some time ago.  Maybe stuff has broken since then.

If you want this resolved, you must post a much more detailed bug report.  The exact code you're using in Arduino (if not exactly one of the library's examples) is a bare minimum.

lenny1972, 2015-06-23:
Paul, I was just trying to compile one of your example (TimeRTC) on IDE 1.6.0 (Arduiono DUE - Programming Port) but the compiler give me the following errors:

C:\Users\471887\Documents\Arduino\libraries\DS1307RTC/DS1307RTC.h:27:18: note: in expansion of macro 'RTC'
 extern DS1307RTC RTC;
                  ^
TimeRTC.pde: In function 'void setup()':
TimeRTC.pde:13:23: error: request for member 'get' in '1074666080u', which is of pointer type 'Rtc*' (maybe you meant to use '->' ?)
TimeRTC.pde:13:26: error: 'setSyncProvider' was not declared in this scope
TimeRTC.pde:14:17: error: 'timeStatus' was not declared in this scope
TimeRTC.pde:14:21: error: 'timeSet' was not declared in this scope
TimeRTC.pde: In function 'void digitalClockDisplay()':
TimeRTC.pde:28:21: error: 'hour' was not declared in this scope
TimeRTC.pde:29:22: error: 'minute' was not declared in this scope
TimeRTC.pde:30:22: error: 'second' was not declared in this scope
TimeRTC.pde:32:20: error: 'day' was not declared in this scope
TimeRTC.pde:34:22: error: 'month' was not declared in this scope
TimeRTC.pde:36:21: error: 'year' was not declared in this scope
Errore durante la compilazione

michaelmargolis, 2015-06-23:
On my Windows 7 PC, TimeRTC compiles without errors or warnings using IDE 1.6.5 and 1.6.2 (I no longer have 1.6.0 installed).
Try a later version of the IDE to see if that is triggering the errors.

PaulStoffregen, 2015-06-23:
Compiles here on Arduino 1.6.5.

[image]

lenny1972, 2015-06-23:
Sorry Paul, it was probably a problem with my installation od the arduino IDE (1.6.0).
I installed the 1.6.5 r2 version of the IDE ad it compiled like flawlessly.

Thanks for you prompt feedback.

Lenny

closed 2015-06-23 by PaulStoffregen