PaulStoffregen/Time #23: Wrong weekday()

bimbo385, 2015-08-16:
The weekday() function returns wrong daynumbers.

If I understand the documentation right, Sunday is 0, Monday is 1, ... and Saturday is 6. 

But yesterday (16.08.2015, Sunday) I tested the library and weekday() gives me a 1, now it's Monday (17.08.2015) and weekday() gives me a 2 for Tuesday... 

I used your library to convert an NTP timestamp to human readable strings on an ESP8266. Date and Time is working correct, so I be certain that this is not an issue with a wrong Unix timestamp or my program.

Can you please check this?
Thank You!

PS: I can give you my sketch if you like. Maybe via e-mail.

DigitalGrowControl, 2016-05-22:
I have a similar problem of weekday is always two days in the future ... I think it has to do with the leap year.

PaulStoffregen, 2016-05-24:
Please post a program which reproduces the problem (using regular Arduino, *not* requiring ESP).  Post a complete program which can be copied into Arduino and used, not merely a code fragment.

Even if this problem is simple to recreate, please understand I have many other commitments.  I really can't put time into this until a complete program is posted.

joshfeliciano, 2016-12-16:
Today is Thursday, and the following, placed in setup(), prints "6" in the Arduino IDE "Serial Monitor":
```
Serial.begin(9600); // Baud rate for communicating with computer serial
while (!Serial); // Wait for serial
Alarm.delay(250);

RTC.read(sys_clk);
tmElements_t tm;
breakTime(makeTime(sys_clk), sys_clk);
Serial.println(tm.Wday);
```

PaulStoffregen, 2017-12-13:
> If I understand the documentation right, Sunday is 0, Monday is 1, ... and Saturday is 6.

The readme says:

    weekday();         // day of the week (1-7), Sunday is day 1

All the code comments I could find agree, Sunday is 1.

closed 2017-12-13 by PaulStoffregen