PaulStoffregen/Time #32: Arduino 1.6.6 breaks Time library

blackketter, 2015-11-07:
There seems to be a new conflict between the ctime time.h and the Arduino Time.h library.  
Reproduce by building TimeTeensy3 example.  This is on Mac, where the filesystem is ignores case, which may be part of the cause.  Verbose build results below.

```
Arduino: 1.6.6 (Mac OS X), TD: 1.26-beta3, Board: "Teensy 3.2 / 3.1, Serial + Keyboard + Mouse + Joystick, 72 MHz optimized, US English"

/Users/dean/Downloads/Arduino-1.6.6.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware "/Users/dean/Downloads/Arduino-1.6.6.app/Contents/Java/hardware" -hardware "/Users/dean/Library/Arduino15/packages" -tools "/Users/dean/Downloads/Arduino-1.6.6.app/Contents/Java/tools-builder" -tools "/Users/dean/Downloads/Arduino-1.6.6.app/Contents/Java/hardware/tools/avr" -tools "/Users/dean/Library/Arduino15/packages" -built-in-libraries "/Users/dean/Downloads/Arduino-1.6.6.app/Contents/Java/libraries" -libraries "/Users/dean/Documents/Arduino/libraries" -fqbn=teensy:avr:teensy31:usb=serialhid,speed=72opt,keys=en-us -ide-version=10606 -build-path "/var/folders/w1/0xxlqytr8xg565001s7pswz80000gn/T/build98fee79b5753e07289d9cb5c4db23e65.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/Users/dean/Documents/Arduino/libraries/TimeLib/examples/TimeTeensy3/TimeTeensy3.ino"
/Users/dean/Downloads/Arduino-1.6.6.app/Contents/Java/arduino-builder -compile -logger=machine -hardware "/Users/dean/Downloads/Arduino-1.6.6.app/Contents/Java/hardware" -hardware "/Users/dean/Library/Arduino15/packages" -tools "/Users/dean/Downloads/Arduino-1.6.6.app/Contents/Java/tools-builder" -tools "/Users/dean/Downloads/Arduino-1.6.6.app/Contents/Java/hardware/tools/avr" -tools "/Users/dean/Library/Arduino15/packages" -built-in-libraries "/Users/dean/Downloads/Arduino-1.6.6.app/Contents/Java/libraries" -libraries "/Users/dean/Documents/Arduino/libraries" -fqbn=teensy:avr:teensy31:usb=serialhid,speed=72opt,keys=en-us -ide-version=10606 -build-path "/var/folders/w1/0xxlqytr8xg565001s7pswz80000gn/T/build98fee79b5753e07289d9cb5c4db23e65.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/Users/dean/Documents/Arduino/libraries/TimeLib/examples/TimeTeensy3/TimeTeensy3.ino"
WARNING: Spurious .git folder in 'Adafruit GFX Library' library
WARNING: Spurious .git folder in 'PID' library
WARNING: Spurious .git folder in 'Audio' library
WARNING: Spurious .git folder in 'FastLED' library
WARNING: Spurious .git folder in 'ILI9341_t3' library
WARNING: Spurious .git folder in 'SparkFun APDS9960 RGB and Gesture Sensor' library
"/Users/dean/Downloads/Arduino-1.6.6.app/Contents/Java/hardware/tools/avr/../arm/bin/arm-none-eabi-g++" -M -MG -MP -x c++ -w -fno-exceptions -felide-constructors -std=gnu++0x -fno-rtti -mthumb -mcpu=cortex-m4 -fsingle-precision-constant -D__MK20DX256__ -DTEENSYDUINO=126 -DARDUINO=10606 -DF_CPU=72000000 -DARDUINO_ARCH_AVR -DUSB_SERIAL_HID -DLAYOUT_US_ENGLISH "-I/Users/dean/Downloads/Arduino-1.6.6.app/Contents/Java/hardware/teensy/avr/cores/teensy3" "/var/folders/w1/0xxlqytr8xg565001s7pswz80000gn/T/build98fee79b5753e07289d9cb5c4db23e65.tmp/sketch/TimeTeensy3.ino.cpp"
"/Users/dean/Downloads/Arduino-1.6.6.app/Contents/Java/hardware/tools/avr/../arm/bin/arm-none-eabi-g++" -M -MG -MP -x c++ -w -fno-exceptions -felide-constructors -std=gnu++0x -fno-rtti -mthumb -mcpu=cortex-m4 -fsingle-precision-constant -D__MK20DX256__ -DTEENSYDUINO=126 -DARDUINO=10606 -DF_CPU=72000000 -DARDUINO_ARCH_AVR -DUSB_SERIAL_HID -DLAYOUT_US_ENGLISH "-I/Users/dean/Downloads/Arduino-1.6.6.app/Contents/Java/hardware/teensy/avr/cores/teensy3" "/var/folders/w1/0xxlqytr8xg565001s7pswz80000gn/T/build98fee79b5753e07289d9cb5c4db23e65.tmp/sketch/TimeTeensy3.ino.cpp"
"/Users/dean/Downloads/Arduino-1.6.6.app/Contents/Java/hardware/tools/avr/../arm/bin/arm-none-eabi-g++" -M -MG -MP -x c++ -w -fno-exceptions -felide-constructors -std=gnu++0x -fno-rtti -mthumb -mcpu=cortex-m4 -fsingle-precision-constant -D__MK20DX256__ -DTEENSYDUINO=126 -DARDUINO=10606 -DF_CPU=72000000 -DARDUINO_ARCH_AVR -DUSB_SERIAL_HID -DLAYOUT_US_ENGLISH "-I/Users/dean/Downloads/Arduino-1.6.6.app/Contents/Java/hardware/teensy/avr/cores/teensy3" "/var/folders/w1/0xxlqytr8xg565001s7pswz80000gn/T/build98fee79b5753e07289d9cb5c4db23e65.tmp/sketch/TimeTeensy3.ino.cpp"
"/Users/dean/Downloads/Arduino-1.6.6.app/Contents/Java/hardware/tools/avr/../arm/bin/arm-none-eabi-g++" -M -MG -MP -x c++ -w -fno-exceptions -felide-constructors -std=gnu++0x -fno-rtti -mthumb -mcpu=cortex-m4 -fsingle-precision-constant -D__MK20DX256__ -DTEENSYDUINO=126 -DARDUINO=10606 -DF_CPU=72000000 -DARDUINO_ARCH_AVR -DUSB_SERIAL_HID -DLAYOUT_US_ENGLISH "-I/Users/dean/Downloads/Arduino-1.6.6.app/Contents/Java/hardware/teensy/avr/cores/teensy3" "/var/folders/w1/0xxlqytr8xg565001s7pswz80000gn/T/build98fee79b5753e07289d9cb5c4db23e65.tmp/sketch/TimeTeensy3.ino.cpp"
"/Users/dean/Downloads/Arduino-1.6.6.app/Contents/Java/hardware/tools/avr/../arm/bin/arm-none-eabi-g++" -M -MG -MP -x c++ -w -fno-exceptions -felide-constructors -std=gnu++0x -fno-rtti -mthumb -mcpu=cortex-m4 -fsingle-precision-constant -D__MK20DX256__ -DTEENSYDUINO=126 -DARDUINO=10606 -DF_CPU=72000000 -DARDUINO_ARCH_AVR -DUSB_SERIAL_HID -DLAYOUT_US_ENGLISH "-I/Users/dean/Downloads/Arduino-1.6.6.app/Contents/Java/hardware/teensy/avr/cores/teensy3" "/var/folders/w1/0xxlqytr8xg565001s7pswz80000gn/T/build98fee79b5753e07289d9cb5c4db23e65.tmp/sketch/TimeTeensy3.ino.cpp"
"/Users/dean/Downloads/Arduino-1.6.6.app/Contents/Java/hardware/tools/avr/../arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w  -g -Wall -ffunction-sections -fdata-sections -MMD -nostdlib -fno-exceptions -felide-constructors -std=gnu++0x -fno-rtti -mthumb -mcpu=cortex-m4 -fsingle-precision-constant -D__MK20DX256__ -DTEENSYDUINO=126 -DARDUINO=10606 -DF_CPU=72000000 -DARDUINO_ARCH_AVR -DUSB_SERIAL_HID -DLAYOUT_US_ENGLISH "-I/Users/dean/Downloads/Arduino-1.6.6.app/Contents/Java/hardware/teensy/avr/cores/teensy3" "/var/folders/w1/0xxlqytr8xg565001s7pswz80000gn/T/build98fee79b5753e07289d9cb5c4db23e65.tmp/sketch/TimeTeensy3.ino.cpp"
"/Users/dean/Downloads/Arduino-1.6.6.app/Contents/Java/tools-builder/ctags/5.8-arduino2/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns "/var/folders/w1/0xxlqytr8xg565001s7pswz80000gn/T/build98fee79b5753e07289d9cb5c4db23e65.tmp/preproc/ctags_target.cpp"
"/Users/dean/Downloads/Arduino-1.6.6.app/Contents/Java/tools-builder/ctags/5.8-arduino2/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns "/var/folders/w1/0xxlqytr8xg565001s7pswz80000gn/T/build98fee79b5753e07289d9cb5c4db23e65.tmp/preproc/ctags_target.cpp"
"/Users/dean/Downloads/Arduino-1.6.6.app/Contents/Java/hardware/tools/avr/../arm/bin/arm-none-eabi-g++" -c -O -g -Wall -ffunction-sections -fdata-sections -MMD -nostdlib -fno-exceptions -felide-constructors -std=gnu++0x -fno-rtti -mthumb -mcpu=cortex-m4 -fsingle-precision-constant -D__MK20DX256__ -DTEENSYDUINO=126 -DARDUINO=10606 -DF_CPU=72000000 -DARDUINO_ARCH_AVR -DUSB_SERIAL_HID -DLAYOUT_US_ENGLISH "-I/Users/dean/Downloads/Arduino-1.6.6.app/Contents/Java/hardware/teensy/avr/cores/teensy3" "/var/folders/w1/0xxlqytr8xg565001s7pswz80000gn/T/build98fee79b5753e07289d9cb5c4db23e65.tmp/sketch/TimeTeensy3.ino.cpp" -o "/var/folders/w1/0xxlqytr8xg565001s7pswz80000gn/T/build98fee79b5753e07289d9cb5c4db23e65.tmp/sketch/TimeTeensy3.ino.cpp.o"
/Users/dean/Documents/Arduino/libraries/TimeLib/examples/TimeTeensy3/TimeTeensy3.ino: In function 'void setup()':
/Users/dean/Documents/Arduino/libraries/TimeLib/examples/TimeTeensy3/TimeTeensy3.ino:11:33: error: 'setSyncProvider' was not declared in this scope
setSyncProvider(getTeensy3Time);
^
/Users/dean/Documents/Arduino/libraries/TimeLib/examples/TimeTeensy3/TimeTeensy3.ino:16:18: error: 'timeStatus' was not declared in this scope
if (timeStatus()!= timeSet) {
^
/Users/dean/Documents/Arduino/libraries/TimeLib/examples/TimeTeensy3/TimeTeensy3.ino:16:22: error: 'timeSet' was not declared in this scope
if (timeStatus()!= timeSet) {
^
/Users/dean/Documents/Arduino/libraries/TimeLib/examples/TimeTeensy3/TimeTeensy3.ino: In function 'void loop()':
/Users/dean/Documents/Arduino/libraries/TimeLib/examples/TimeTeensy3/TimeTeensy3.ino:28:16: error: 'setTime' was not declared in this scope
setTime(t);
^
/Users/dean/Documents/Arduino/libraries/TimeLib/examples/TimeTeensy3/TimeTeensy3.ino: In function 'void digitalClockDisplay()':
[... cut ...]
/Users/dean/Documents/Arduino/libraries/TimeLib/examples/TimeTeensy3/TimeTeensy3.ino:45:21: error: 'year' was not declared in this scope
Serial.print(year());
^
/Users/dean/Documents/Arduino/libraries/TimeLib/examples/TimeTeensy3/TimeTeensy3.ino: In function 'long unsigned int processSyncMessage()':
/Users/dean/Documents/Arduino/libraries/TimeLib/examples/TimeTeensy3/TimeTeensy3.ino:61:29: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
if(Serial.find(TIME_HEADER)) {
^
exit status 1
'year' was not declared in this scope
```

gmag11, 2015-12-26:
Same happens with ESP8266 Arduino environment. I had to rename Time.h to Time2.h in this library and all its references. After that it compiles ok, but on every library update I have to do it again.

PaulStoffregen, 2015-12-26:
Fixed by https://github.com/PaulStoffregen/Time/commit/26cd0c85095d5dbfffded0831abb2f7abaad0469

Simply update to the latest code.

closed 2015-12-26 by PaulStoffregen