PaulStoffregen/Time #33: makeTime function

Ben20082010, 2015-11-22:
Hi,
I had some problem with this function, When I ask to makeTime what is 0day 0month 0year, it return “4294880896”, which is “6:28:16 2-6-2106”. I had been try to find out the reason for last past hours and couldn't tell why. Please help me

Here is my code.
#include<Time.h>
void setup() {
  Serial.begin(9600);
}

void loop() {
  setTime(sync());
  Serial.print(now());
  Serial.print("   ");
  Serial.print(hour());
  Serial.print(':');
  Serial.print(minute());
  Serial.print(':');
  Serial.print(second());
  Serial.print(' ');
  Serial.print(month());
  Serial.print('-');
  Serial.print(day());
  Serial.print('-');
  Serial.println(year());
}
time_t sync() {
  tmElements_t tm;
  tm.Hour = 0;
  tm.Minute = 0;
  tm.Second = 0;
  tm.Day = 0;
  tm.Month = 0;
  tm.Year = 0;
  return makeTime(tm);
}

Thank you

blackketter, 2015-11-22:
The tmElements_t data structure you are creating is not valid.  

Keep in mind that the years in a tmElements structure are since 1970 and that the months start at 1 (for January) as do the days.

I hope this helps.

Ben20082010, 2015-11-22:
Thanks for your help.
But I don't think this would be a problem as the time_t means how many seconds  has passed after 1/1/1970.
Also, When I change the day to 1, month to 1, year to 1970, it returns 1322155904 which is 17:31:44 11-24-2011, it just totally doesn't make sense.

Ben20082010, 2015-11-22:
Would you mind try this or similar code on your Arduino? As it's possible because something wrong with my IDE on the computer. But it didn't help even after I reinstalled this Time library.

PaulStoffregen, 2015-11-23:
Indeed your error is use of 0 for tm.Day and tm.Month.

I just tested this code:

    #include <Time.h>
    void setup() {
      Serial.begin(9600);
    }

    void loop() {
      setTime(sync());
      Serial.print(now());
      Serial.print(" ");
      Serial.print(hour());
      Serial.print(':');
      Serial.print(minute());
      Serial.print(':');
      Serial.print(second());
      Serial.print(' ');
      Serial.print(month());
      Serial.print('-');
      Serial.print(day());
      Serial.print('-');
      Serial.println(year());
      delay(500);
    }
    time_t sync() {
      tmElements_t tm;
      tm.Hour = 0;
      tm.Minute = 0;
      tm.Second = 0;
      tm.Day = 1;
      tm.Month = 1;
      tm.Year = 0;
      return makeTime(tm);
    }

This is the result in the serial monitor:

    0 0:0:0 0-0-1970
    0 0:0:0 0-0-1970
    0 0:0:0 0-0-1970
    0 0:0:0 0-0-1970
    0 0:0:0 0-0-1970

Time is working properly.  You simply used 0 in those fields which begin at 1.

closed 2015-11-23 by PaulStoffregen