PaulStoffregen/Time #35: Strange problem using ESP8266 sample

lunanigra, 2016-01-23:
Hello, I'm trying to use your sample sketch with an ESP8266 chip. However even when using your code exactly as it is I'm running on an error...

```
/Users/me/Documents/Arduino/ESP8266_NTP/ESP8266_NTP.ino: In function 'void setup()':
ESP8266_NTP:54: error: 'getNtpTime' was not declared in this scope
   setSyncProvider(getNtpTime);
                   ^
/Users/me/Documents/Arduino/ESP8266_NTP/ESP8266_NTP.ino: In function 'void loop()':
ESP8266_NTP:64: error: 'digitalClockDisplay' was not declared in this scope
       digitalClockDisplay();  
                           ^
/Users/me/Documents/Arduino/ESP8266_NTP/ESP8266_NTP.ino: In function 'void digitalClockDisplay()':
ESP8266_NTP:72: error: 'printDigits' was not declared in this scope
   printDigits(minute());
                       ^
/Users/me/Documents/Arduino/ESP8266_NTP/ESP8266_NTP.ino: In function 'time_t getNtpTime()':
ESP8266_NTP:102: error: 'sendNTPpacket' was not declared in this scope
   sendNTPpacket(timeServer);
                           ^
exit status 1
'getNtpTime' was not declared in this scope
```

Your code looks totally fine! So, it's quite strange... Compiling the original Arduino sketch (for a real Arduino controller) works fine as well... Then I changed the order of your functions (e.g. putting setup/loop) to the end. After that no compiler issue.

Any idea what is going wrong? I assume that the sketch works for you. Making it even stranger...
Or is there any flag handling function ordering?

Thanks, JC

chaeplin, 2016-01-23:
add this after #include <WiFiUdp.h>
```
time_t getNtpTime();
void digitalClockDisplay();
void printDigits(int digits);
void sendNTPpacket(IPAddress &address);
```

Known issue in esp8266
https://github.com/esp8266/Arduino/issues/1472

PaulStoffregen, 2016-01-23:
Since this isn't a bug in the Time library, I'm closing this issue.  It really needs to be fixed in the tools.

closed 2016-01-23 by PaulStoffregen

lunanigra, 2016-01-23:
Thanks for your help! Works :-)