PaulStoffregen/Time #37: Compile errors in Arduino 1.6.7

gerardwr, 2016-02-12:
Hi Paul,

The example TimeNTP_ESP8226Wifi works fine in Arduino 1.6.4, thanks for sharing!

In Arduino 1.6.7 I get plenty compile errors, as I understand caused by the (now) required "Forward Declaration".

If I shuffle declarations and function around, the sketch compiles and upload OK in Arduino 1.6.7.

Thanks,
Gerard.

PaulStoffregen, 2016-02-17:
Please submit a pull request.

I do not have an ESP board set up for testing, so I'm depending on ESP users to submit pull requests to keep the ESP example updated.

gerardwr, 2016-02-17:
Alas, I've got no experience in submitting a pull request, but I've attached my modified sketch that compiles in Arduino 1.6.7 and runs on a Wemos D1 ESP8266 board..
[Uploading TimeNTP_ESP8266WiFi.ino…]()

closed 2016-02-17 by gerardwr