PaulStoffregen/Time #39: IDE 1.6.7 build problem with timelib.h

Defragster, 2016-03-05:
I get this error in the IDE when I complete an UPLOAD compile once - then try to upload/compile a second time:

> Using library Time at version 1.5 in folder: i:\tcode\libraries\Time 
> GetFileAttributesEx i:\tcode\libraries\Time\DateStrings.cpp i:\tcode\libraries\Time\Time.h: The filename, directory name, or volume label syntax is incorrect.
> Error compiling

This is for an ESP8266 (v2.0 and now on v2.1) system on Windows 7.  

I've found a workaround where I open the GitHub .ZIP file and copy/overwrite the problem files and then recompile. This works but is a pain to hold both directories open and perform the copy after each failure. It does not work to just copy the files from a second directory as I tried it - but only from the open ZIP.

This must be an IDE issue holding a strange attachment to the files I've never seen before.

This library brought in externally for this library as specified where that owner has not seen this:
[](https://github.com/nailbuster/myWebServer)

PaulStoffregen, 2016-03-05:
This problem was fixed by version 1.5.  Look at the library.properties file.  Is your copy of Time 1.4 or 1.5?

Defragster, 2016-03-05:
Note above: Using library Time at version 1.5 in folder: i:\tcode\libraries\Time 

**// "I:\tCode\libraries\Time\library.properties"**
> name=Time
> version=1.5
> author=Michael Margolis
> maintainer=Paul Stoffregen

PaulStoffregen, 2016-03-05:
Oh, I think I see what might be going on here.  Let's try a simple fix and see if it helps...

Defragster, 2016-03-05:
ready ...

PaulStoffregen, 2016-03-05:
Ok, does this help?
https://github.com/PaulStoffregen/Time/commit/0bfe218b56adce50a8d5796e24a9692b99c4a802

Defragster, 2016-03-05:
GitHub: Forked, copied to desktop, files copied.
UPLOAD ( OTA to ESP ) , good
UPLOAD ( OTA to ESP ) , good
UPLOAD ( OTA to ESP ) , good

Those file edit fixed it.
Thx Paul.

closed 2016-03-05 by Defragster

pieman64, 2016-04-15:
@PaulStoffregen I have the same problem with IDE 1.6.8 amd Time 1.5.0.
I moved some of my files to another machine today and I have tried https://github.com/PaulStoffregen/Time/commit/0bfe218b56adce50a8d5796e24a9692b99c4a802 but it doesn't seem to fix it.

I think I have a similar fix to @Defragster.

If I open the IDE and sketch it will compile without error but if I try to compile a second time I get:

**GetFileAttributesEx D:\Arduino\libraries\Time\Time.h D:\Arduino\libraries\Time\TimeLib.h: The filename, directory name or volume label syntax is incorrect.
Error compiling for board Generic ESP8266 Module.**

Generic ESP8266 on Windows 8.1 and ESP8266 2.1.0.

Defragster, 2016-04-15:
#include <TimeLib.h>

At the top of my INO file after the fix - with the new (sketchbook)\libraries\TIME fixed it for me.
With ESP8266 2.1.0 and IDE 1.6.8 on Win 7.