PaulStoffregen/Time #41: adjustTime() not compatible with sync provider.

bperrybap, 2016-03-21:


TheAustrian, 2016-04-07:
I don't know if you found the solution already, but the getNtpTime()-function is the place where you could add/subtract any offset you need and have it applied every time you fetch new data. Just adjust the line:

return secsSince1900 - 2208988800UL + timeZone * SECS_PER_HOUR;
to
return secsSince1900 - 2208988800UL + timeZone * SECS_PER_HOUR + wantedOffset;

bperrybap, 2016-04-07:
@TheAustrian modifying an individual sync provider can work around the issue for that individual sync provider, but it doesn't solve the problem and it requires adding some additional APIs outside the Time library to set the wantedOffset that will be specify to only that sync provider.
And even after doing all that, the issue with using adjustTime() still exists.

The real solution is to update the Time library to provide an offset capability so that it works consistently with or without using a sync provider.

TheAustrian, 2016-04-07:
Ok, though the function is the same for any NTP server you use and I don't see what else would change the value of an NTP-dependent time variable. Also you specifically asked for something that adds the offset on sync, which the line I proposed does.

I can't say anything about another implementation, its value for the general public and if it would be much work to implement or not...

bperrybap, 2016-04-07:
The Time library is not specific to a sync provider, and sync provider code is not part of the TIme library, any work around solution like shown above would be outside the scope of the Time library.

There are actually two issues here:
- Incorrectly using adjustTime() when using a sync provider, as it can't be used with a sync provider.
- Providing a time offset capability.

The two are separate and distinct.

There is a tiny patch I proposed in the Arduino thread that is a work around for users that incorrectly use adjustTime() with a sync provider that will solve the issue they run into in the typical misuse case. This doesn't really solve anything but does allow quite a bit of broken code to actually do what the author thought it should it do.

Solving how to provide a time offset capability API extension is more complicated.

A time offset capability has nothing to do with NTP, the same issues exist will all sync providers.
And while you can do a work around hack like you showed above, which also can be done in all other sync providers, it doesn't work correctly for all uses, since from a technical perspective there are problems when the sync provider messes with the actual time epoch.
That isn't the way timezone offsets should to work when using an epoch.

There are actually two ways time offsets can be handled in the Time library.
- As a hack to trick the Time library into using an epoch value that has been offset to localtime
(While this works for some simple use cases, it is not ideal and has issues since it corrupts the actual time)
- add a offset capability so that the offsets are only added/used when needed/desired.
(done correctly it should work when setting the time)

The sync provider work around for NTP you showed above falls into the first method above as it works for some use cases, but has issues with others.

One compromise solution may be to do something similar to what aarg proposed in that arduino thread:
```
int hour = hour( UTCtoLocal(now()) );
```
While aarg's proposal is easy fairly easy to implement, I think that there are other  better ways to handle it including supporting the offset for time setting.

When using an epoch correctly, a time shift offset should be used to offset an epoch time value to the local timzone only during the conversion process from epoch to human form when calculating the human readable values like hr, min, etc...
A time offset capability should NEVER modify the actual epoch.

This ensures that the epoch time never changes regardless of the offset.
And this is very important to ensure time consistency.

The entire point of an epoch is that it is based on elapsed time from a specific point in time.
The problem with adding an offset to the sync provider time or having the sync provider add an offset is that it is now corrupting the actual time since it is no longer using a consistent epoch.

Maintaining a proper consistent known epoch is critical as that is how time is actually tracked when using an epoch.
Without a consistent epoch, all kinds of things start to break since you no longer know what time the epoch represents.
There is no way to do timestamping since the timestamp is the epoch and if you break down the epoch later back into human readible form, there is no way to know how to do that since you don't know the offset that was added into the timestamp/epoch value.
Things like automatic summertime or Daylight savings adjustments can become problematic if the epoch is not consistent.
Things that use alarms based on the epoch don't work correctly as the actual epoch is changing which alters when the alarm will trigger.

All that said, some users don't care and are ok with corrupting their epoch as long as the human readable time looks ok . They are not concerned with summertime/Daylight savings adjustments, timestamps, or other uses that require time consistency.
Those users will happy with the solution that modifies the epoch.

But like I said, a real solution needs to work for all use cases, all sync providers, and the API needs to be part of the Time library so it is handled consistently regardless of whether a sync provider is used or which sync provider is used.
Once the Time library supports time offsets, it works for all users, all use cases, and all sync providers.

Mike4U, 2020-04-27:
This and issue #61 are because adjustTime(long adjustment) in Time.cpp does not contain:
         nextSyncTime = (uint32_t)t + syncInterval;
         Status = timeSet;
a fix is don't use adjustTime() use:
         setTime(now()+adjustment);