PaulStoffregen/Time #43: Sommerzeit and Winterzeit

holgerlembke, 2016-07-15:
Paul, how to handle Sommerzeit and Winterzeit (this stuff we do in Germany twice a year)?

``boolean isgermansommerzeit() {
  // http://www.mikrocontroller.net/topic/25069
  // http://www.mikrocontroller.net/attachment/8391/TIME.C

  // volle Monate mit Winterzeit
  if ( (month() < 3) || (month() > 10) ) {   // month 1, 2, 11, 12
    return false;         // -> Winterzeit
  }

  byte wday = weekday();

  if (day() - wday >= 25 && (wday || hour() >= 2) ) { // after last Sunday 2:00
    if ( month() == 10 )       // October -> Winter
      return false;
  } else {         // before last Sunday 2:00
    if ( month() == 3 )       // March -> Winter
      return false;
  }

  return true;
}

time_t now() {
	// calculate number of seconds passed since last call to now()
  while (millis() - prevMillis >= 1000) {
		// millis() and prevMillis are both unsigned ints thus the subtraction will always be the absolute value of the difference
    sysTime++;
    prevMillis += 1000;	
#ifdef TIME_DRIFT_INFO
    sysUnsyncedTime++; // this can be compared to the synced time to measure long term drift     
#endif
  }
  if (nextSyncTime <= sysTime) {
    if (getTimePtr != 0) {
      time_t t = getTimePtr();
      if (t != 0) {
        setTime(t);
		if (isgermansommerzeit()) {
	      t+=60*60;
          setTime(t);
		}
      } else {
        nextSyncTime = sysTime + syncInterval;
        Status = (Status == timeNotSet) ?  timeNotSet : timeNeedsSync;
      }
    }
  }  
  return (time_t)sysTime;
}

magicsunday, 2017-03-18:
Its called DST (daylight saving time).

blackketter, 2017-03-18:
This library works well for me for both time zones and DST: https://github.com/JChristensen/Timezone

holgerlembke, 2017-03-18:
I see my inital question with some shame, because I was assumed some things that are not so obvious than I thought...

And @magicsunday already added that into his answer: how to handle the different DSTs inside the component? I thought that Paul perhaps might add some prototypes that fit into his concept (because he knows best) and third parties could implement their DST routine into it. So over time there might be a pool of working DST routines.