PaulStoffregen/Time #46: Weekday is 2 today - Monday

pieman64, 2016-08-08:
Weekday() is showing as 2 today buy your notes indicate Sunday is 0 so surely Monday should be 1.

Please advise.

ljruiten, 2016-10-28:
I have had a quick look through the source code and at the weekday function there is a little comment that says: **// sunday is day 1**, so the documentation is wrong. Sunday comes first in the index, but it is indexed at 1 and not at 0. 

Today is Friday and that returns 6. I then tried weekday(now() + 86400); (the amount of seconds in a day) and it started returning 7. And weekday(now() + 172800); returns 1. So sunday = 1 and saturday = 7.

So to summarize if you want to have it 0 indexed (like I did) just subtract 1. This should either be updated in the documentation or fixed in the source code though, because in any case one of them is wrong... I believe this is a bug in the source, because it usually is the convention to return 0 for a sunday.

smz, 2017-03-10:
I'm an "intruder" here as I'm not currently using this library, but I think 0 = Sunday based weekdays (_as used in the standard C time library_) makes sense as you can perform "modulo 7" modular arithmetic on "number of days"

PaulStoffregen, 2017-03-10:
The readme says "day of the week (1-7), Sunday is day 1".  The Arduino Playground documentation says "Day of the week, Sunday is day 1".

I've edited the playground page to now say "Day of the week (1-7), Sunday is day 1".

Sorry, this library is not going to make an incompatible change to 0-6 that will break so many existing programs.  It's just not going to happen.  Please stop asking.

If there is any incorrect documentation, please tell me specifically where you see it?  @pieman64 you wrote "but your notes indicate Sunday is 0".  Where do you see this info?

PaulStoffregen, 2017-03-10:
Likewise, month() is not changing to zero based.  @michaelmargolis (the original author of this library) and I talked about the API in person.  This is a very well established library, since the very early days of Arduino, now used by many thousands of programs.

We are **not going to make incompatible API changes**.  The answer is no.

smz, 2017-03-11:
Thanks, I perfectly understand the rationale behind your choices (_and agree, although this surely is not much important!_)

closed 2017-12-13 by PaulStoffregen