PaulStoffregen/Time #54: Can't install into arduino IDE 1.6.12

ebeyer, 2016-09-23:
The installer refuses to install libraries into the latest arduino IDE (v1.6.12, as of this post).  It supports v1.6.11.

PaulStoffregen, 2016-09-24:
[image]

ebeyer, 2016-09-24:
But how does that help me _now_?   :)

Seriously, though.  I'm looking forward to using your library for this really cool project (https://learn.adafruit.com/flora-geo-watch/overview), so thanks for all of your work.
EB

PaulStoffregen, 2016-09-24:
Well, you click that link, then on the page with the beta versions, you download the one you need (depending on your operating system).  Then run the installer you just downloaded.  It will recognize and install into Arduino 1.6.12.  Then run 1.6.12, select Teensy from the Tools - Boards menu, and use it normally.

How can I improve the website to make this clearer?

ebeyer, 2016-09-24:
I'm really sorry.  Your web site is plenty clear.  But I got to the downloader by way of the Adafruit project I linked you, which takes you to its github page, which directs you to https://www.pjrc.com/teensy/td_libs_Time.html.  If you forge ahead to the download link, as I did, and not to the Github one, you get the non-beta version.  I'll give the beta a go and let you know if there are issues.

Thanks again.

closed 2016-09-24 by ebeyer