PaulStoffregen/Time #58: change the synchronization time and make continuous requests in case you can not find server response

arlaor, 2016-10-19:
Hi.
as indicated by the question as I can make time to upgrade to Internet is not made once every 5 minutes now makes the code [TimeNTP_ESP8266WiFi.ino](https://github.com/PaulStoffregen/Time/blob/master/examples/TimeNTP_ESP8266WiFi/TimeNTP_ESP8266WiFi.ino) . I want to make the initial sincronization continuously until you connect and synchronize again 24 hours later.
What are the lines of code to be modified?

blackketter, 2016-10-19:
That example is instructive, but check out https://github.com/arduino-libraries/NTPClient 

I'm using that with the following code:
```
WiFiUDP ntpUDP;
NTPClient timeClient(ntpUDP);

time_t ntpSyncProvider() {
  return timeClient.getEpochTime();
}
```

and then in setup():  
```
  timeClient.begin();
  timeClient.setTimeOffset(timeZoneOffset);

  setSyncProvider(&ntpSyncProvider);
  setSyncInterval(5);
```

and in loop():

```
  timeClient.update();
```

arlaor, 2016-10-19:
Hi @blackketter in the related example I not find the parameters used for the hours, minutes and seconds, everything happens outside the code.
I want to use the example for my project that is turn on at certain days, every hour of the day, a relay and keep it on for a few minutes. So I need to manipulate the variables of time to do, but they do not appear in the code example.
I have reviewed the library [NTPClient.h](https://github.com/arduino-libraries/NTPClient/blob/master/NTPClient.h) but neither appear.

blackketter, 2016-10-19:
If you use my sample code above you can have the Time library use the NTPClient to synchronize the time and then the Time library functions to figure out what day or time it is now.

Beyond this, your question isn't really a bug for the Time library and so should be asked in another forum somewhere.

arlaor, 2016-10-20:
unfortunately I could not make timelib.h or time.h library assume the values of year month day hour minutes seconds.
forgiveness for having overlooked the forum rules.
I appreciate your cooperation.
I tried to follow some other example that fits my needs

closed 2016-10-21 by arlaor