PaulStoffregen/Time #60: A way to initiate / force manual synchronization?

mikekgr, 2016-11-12:
Dear Sir,
first of all I really want to say a big thanks to you for the time library.

I need your help though, I am trying to code for a NTP based project, with automatic timezone update / correction etc using time library of course.
For this reason I need to have an "on demand" way to achieve the same result as ex setSyncInterval(120) command achieve but without to have to wait for the next timed sync.
What do you suggest, is there any way to do this? Is your library has a method to triggering this sync when I want independent from the auto sync provided by setSyncInterval?

Thanks and Best Regards,
Mike Kranidis

blackketter, 2016-11-12:
You could call setSyncProvider() with your existing provider function to force a resync as a side effect.  I'd prefer a new function like:
```
void syncNow() {
  nextSyncTime = 0;
  now();
}
```
I'd be happy to submit a patch for this.

mikekgr, 2016-11-12:
Dear blackketter,
sorry I am not very sure that I understood your suggestion. You means that a simple call like this (in my case) `setSyncProvider(getNtpTime);` will do the trick? 

Thanks and Best Regards,
Mike Kranidis

P.S. I am not sure I pick up the functionality of your void syncNow() function tho

blackketter, 2016-11-12:
> sorry I am not very sure that I understood your suggestion. You means that a simple call like this (in my case) setSyncProvider(getNtpTime); will do the trick?

Yes, just call setSyncProvider() with your time function again.

>P.S. I am not sure I pick up the functionality of your void syncNow() function 

That is a suggestion for a new Time API for explicitly syncing immediately

mikekgr, 2016-11-12:
Dear blackketter,
yes I tested and it worked like a charm!!! Thank you man, you make my day!

Thanks and Best Regards,
Mike Kranidis