PaulStoffregen/Time #63: Issue with setTime and adjustTime when calling hour() immediatly after.

Pearo, 2016-12-20:
to recreate:
make a call to setTime();
make a call to adjustTime(timezoneCorrection);
make a call to hour();

occasionally hour() will return the time not adjusted for the timezone correction

fix:
change refreshCache to force refresh (sorry code markup appears to not be working for me)

void refreshCache(time_t t, bool bForce = false) {
  if (t != cacheTime) {
    breakTime(t, tm); 
    cacheTime = t; 
  }
}

add new call to refreshCache to both setTime() and adjustTime()

void setTime(int hr,int min,int sec,int dy, int mnth, int yr){
 // year can be given as full four digit year or two digts (2010 or 10 for 2010);  
 //it is converted to years since 1970
  if( yr > 99)
      yr = yr - 1970;
  else
      yr += 30;  
  tm.Year = yr;
  tm.Month = mnth;
  tm.Day = dy;
  tm.Hour = hr;
  tm.Minute = min;
  tm.Second = sec;
  setTime(makeTime(tm));
  
  refreshCache(sysTime,true);
}

void adjustTime(long adjustment) {
  sysTime += adjustment;
  
  refreshCache(sysTime,true);
}

Eheran1, 2020-05-09:
Thank you, fixed it for me!
(after finding out that the standard arduino TimeLib.h is actually not from Paul Stoffregen but instead from Michael Margolis, but Paul is the maintainer and I have that library twice due to my Teensy... at the end I corrected this in both librarys... what a mess!)

Im updating the time 10x per Second. Thus I had kind-off the same problem. The time zone correction is only done once per second, leaving the other 9 values wrong. Example (tenth of seconds at the end):
`hh:mm:ss.s
20:42:17.0
18:42:17.1
18:42:17.2
18:42:17.3
18:42:17.4
18:42:17.5
18:42:17.6
18:42:17.7
18:42:17.8
18:42:17.9
20:42:18.0`

I also get the same problem as you [occasionally hour() will return the time not adjusted for the timezone correction]. So calling the adjust once per second instead of 10x per second doesnt help there, every ~30min the problem still popps up for one second.

afch, 2020-07-09:
I did a little differently, added a function call "breakTime" into "adjustTime":

void adjustTime(long adjustment) {
  sysTime += adjustment;
  breakTime(now(), tm); // GRA & AFCH
}