PaulStoffregen/Time #69: Found a strange delay

Bewear, 2017-03-18:
I'm using TimeLib.h and TimeAlarms.h for a clock with alarms, the sync is made through NTP server once everyday and I noticed that every 30 minutes now() loses one second (I'm not using now() but hour(), minute() and second()). I've tested my clock for 7 hours, the time at the beginning was the same my smartphone had and after those 7 hours Arduino lost 14 seconds. Is that a power issue? I'm using the USB port, an LCD screen, 3 buttons, a buzzer, a PIR and WiFI Shield.

gsexton, 2017-03-18:
This is just the inaccuracy of the resonator on the Arduino. You need to either adjust for drift, or sync more frequently. You could also consider using a DS3231 Clock chip. You could use NTP to periodically set the DS3231, and read it. To adjust for drift, use NTP to sync the clock. 24 or so hours later, look at system elapsed time, and the NTP elapsed time. Calculated an offset and then any time you're reading the local clock, multiple the offset by the time since last sync via NTP.

I'm in the process of building a WWVB radio-controlled clock. Looking at my unit (an Inland UNO R3), I lose 55.145 seconds per day, for a drift rate of -6.40e-4. Your drift rate is pretty similar.

Bewear, 2017-03-18:
I have a RTC bit I'm already using all of the pins (WiFi Shield takes 7 pins for communication and the Uno only got 12 pins). Probably I'll resync every 30 minutes or even 10 minutes, or use a Mega.

Bewear, 2017-03-18:
But wait, I'll get a drift even with an external clock?

closed 2017-03-20 by Bewear