PaulStoffregen/Time #74: Time.h conflicts with time.h on case sensitive filesystems

sandeepmistry, 2017-05-10:
Hi @PaulStoffregen,

Some Arduino users are trying to use this library with libraries like [RTCZero](https://github.com/arduino-libraries/RTCZero) that `#include <time.h>` on case sensitive filesystems like macOS and older Windows. Here's a related issue: https://github.com/arduino-libraries/RTCZero/issues/28

The compiler ends up including `Time.h` instead of the POSIX <time.h> inside the RTCZero library which causes compile errors. Would you be open to renaming `Time.h` in the library to avoid conflicts?

PaulStoffregen, 2017-05-10:
https://github.com/PaulStoffregen/Time/issues/32

https://github.com/PaulStoffregen/Time/issues/39

PaulStoffregen, 2017-05-10:
TL;DR = already fixed.  Include TimeLib.h, and don't use very old copies of this lib.  Simple.

PaulStoffregen, 2017-05-10:
Ideally, find whatever old wiki or other web pages these users are seeing and update them!

sandeepmistry, 2017-05-10:
Well, maybe I should propose removing "Time.h" from this repo instead.

As soon as a sketch include "TimeLib.h", the "Time.h" file in this repo will in the header search path and conflict with the POSIX "time.h" includes.

PaulStoffregen, 2017-05-10:
This library aims to preserve compatibility with older versions of Arduino.

closed 2017-12-13 by PaulStoffregen

Niek, 2019-11-08:
I agree with @sandeepmistry - removing the Time.h file (in at least the git version) would be the best.