PaulStoffregen/Time #78: Missing license

derSuessmann, 2017-07-25:
The library does not contain any information on the type of license. This makes it legally unusable (at least it is in my opinion under German copyright law).
I found only the LGPL in Time.cpp.
Please add an open source license. A LICENSE file in the repository would clarify the license for the user. Best practise seems to add a license comment in each source file.

michaelmargolis, 2017-07-25:
Paul, all of the files written by me are under the GNU Lesser General Public License. If the files you have added can also use the same license then feel free to take use the license information in the time.cpp file to cover all files in this library.

adiazulay, 2020-10-14:
I'm using some of the code from the TimeNTP_ESP8266WiFi in an MIT licensed sample project.

Are the examples also licensed under LGPL?