PaulStoffregen/Time #97: Cannot connect to NTP using Arduino WiFi101 WiFi.getTime() Function

BonShaw1, 2018-03-28:
TimeLib interferes with communication with NTP server when using Arduino WiFi.getTime() function. Removal of the Time library restores full function to WiFi.getTime(). Since it is not possible to redirect the WiFi.getTime() function I cannot determine if the bug is limited to a particular NTP server.

All calls to the WiFi.getTime() function return 0.

I found that removing TimeLib and recompiling did not remove the issue. Closing the Arduino IDE and restarting is needed to restore the function of WiFi.getTime().

### Hardware & Software

Board Adafruit Feather ATSAM21D with WINC1500 WiFi module
Shields / modules used - none
Arduino IDE version 1.8.5
Teensyduino version (if using Teensy)
Version info & package name (from Tools > Boards > Board Manager) Adafruit SAMD 1.0.22
Operating system & version Win 10, 64 bit
Any other software or hardware? No.

```
#include <SPI.h>
#include <SD.h>
#include <WiFi101.h>
//#include "TimeLib.h" //various time manipulation functions

char ssid[] = "SSID";        // your network SSID (name)
char pass[] = "pswd";    // your network password (use for WPA, or use as key for WEP)

int status = WL_IDLE_STATUS;

void setup() {
  WiFi.setPins(8,7,4,2);
  Serial.begin(115200);
  while (!Serial);

  // check if the WiFi module works
  if (WiFi.status() == WL_NO_SHIELD) {
    Serial.println("WiFi shield not present");
    // don't continue:
    while (true);
  }

  // attempt to connect to WiFi network:
  while ( status != WL_CONNECTED) {
    Serial.print("Attempting to connect to SSID: ");
    Serial.println(ssid);
    status = WiFi.begin(ssid, pass);

    // wait 5 seconds for connection:
    delay(5000);
  }
  unsigned long epoch;
  int numberOfTries = 0, maxTries = 6;
  do {
    epoch = WiFi.getTime();
    numberOfTries++;
  }
  while ((epoch == 0) && (numberOfTries <= maxTries));

  if (numberOfTries > maxTries) {
    Serial.print("NTP unreachable!!");
    while (1);
  }
  else {
    Serial.print("Epoch received: ");
    Serial.println(epoch);
    Serial.println();
  }
  while (1);
}

  void loop(){
  }
```

### Errors or Incorrect Output

When TimeLib is included in the build the result is "Attempting to connect to SSID: "xxxxxxxxx"
NTP unreachable!!"

wisprer, 2018-08-08:
I can confirm experiencing this issue, and the even worse part of it is that the issue persists until the affected IDE window is closed and re-opened. I will also add that a separate IDE window loaded and ran the sketch fine while the affected window was open.

BonShaw1, 2018-08-28:
I've been pursuing a workaround for this problem since my first post on it. My workaround during a session of editing and compiling is to comment out the reference to the Timelib.h library as well as any use of it in my program. I then compile the remaining code and receive no errors. I then restore the Timelib.h library reference as well as any instructions using it and recompile. Now everything works as it should. It's an ugly workaround but it works every time until another more elegant fix is made. I sure seems like a linker problem. Nothing else has changed since my post about this problem in March '18

To avoid having to remove my lines of code I use ifdef to remove and restore the offending code:

#define PaulStoffegrenTime   // Comment out this line for first compilation.

#ifdef  PaulStoffegrenTime
  #include "TimeLib.h"
#endif

#ifdef  PaulStoffegrenTime
    int TmpYear = year(t);
    int TmpMonth = month(t);
    int TmpDate = day(t);
    int TmpHour = (epoch  % 86400L) / 3600;
    int TmpMinute = (epoch  % 3600) / 60;
    int TmpSecond = epoch % 60;
#endif

ductsoup, 2018-11-12:
I can confirm this issue using an Adafruit Feather M0 WiFi - ATSAMD21 + ATWINC1500. Not sure why this is but if I drop the TimeLib.h and do that another way it's perfectly stable and works great.

S2Doc, 2019-08-21:
I am using the Feather M0 with WINC 1500 and do not have the issue noted above.  I have, however, noted that it not infrequently requires 10-15 calls to getTime() before a response is received.  I think this may be expected with the NTP system; when I use other NTP code to send and receive NTP packets independently of the getTime() function, the same thing happens.

What has worked for me is to try a few times.  If that doesn't get a response, do something else, then come back and try again.  Not pretty, but it has worked well in devices running for over a year now.