PaulStoffregen/Time #101: how the hell you install timelib.h

doit4fun, 2018-07-16:
how do you install timelib.h?

Please use this form only to report code defects or bugs.

For any question, even questions directly pertaining to this code, post your question on the forums related to the board you are using.

Arduino: forum.arduino.cc
Teensy: forum.pjrc.com
ESP8266: www.esp8266.com
ESP32: www.esp32.com
Adafruit Feather/Metro/Trinket: forums.adafruit.com
Particle Photon: community.particle.io

If you are experiencing trouble but not certain of the cause, or need help using this code, ask on the appropriate forum.  This is not the place to ask for support or help, even directly related to this code.  Only use this form you are certain you have discovered a defect in this code!

Please verify the problem occurs when using the very latest version, using the newest version of Arduino and any other related software.

----------------------------- Remove above -----------------------------

### Description

Describe your problem.

### Steps To Reproduce Problem

Please give detailed instructions needed for anyone to attempt to reproduce the problem.

### Hardware & Software

Board
Shields / modules used
Arduino IDE version
Teensyduino version (if using Teensy)
Version info & package name (from Tools > Boards > Board Manager)
Operating system & version
Any other software or hardware?

### Arduino Sketch

```cpp
// Change the code below by your sketch (please try to give the smallest code which demonstrates the problem)
#include <Arduino.h>

// libraries: give links/details so anyone can compile your code for the same result

void setup() {
}

void loop() {
}
```

### Errors or Incorrect Output

If you see any errors or incorrect output, please show it here.  Please use copy & paste to give an exact copy of the message.  Details matter, so please show (not merely describe) the actual message or error exactly as it appears.

PaulStoffregen, 2018-07-17:
> how do you install timelib.h?

https://www.arduino.cc/en/Guide/Libraries

closed 2018-07-17 by PaulStoffregen

nick-youngblut, 2018-12-30:
Sorry to comment on an old thread, but I am very new to arduino programming, and I too found the install very confusing for the following reasons:

1. The library manager (at least for Arduino IDE v 1.9.0-beta) does not include a "Time" or "TimeLib" library
1. This repo is called "Time", not "TimeLib"
1. [https://www.arduino.cc/en/Guide/Libraries](https://www.arduino.cc/en/Guide/Libraries) does not describe how to install from a GitHub repo, so especially for users not accustomed to cloning repos, there are a couple of steps that the user might not be aware of. 

I'm assuming the proper way to install this library is:

1. clone the repo
1. compress the cloned repo as a *.zip
1. go to the Arduino IDE and import the library as a zip file

At least, that's how I installed the library. Any reason why the Arduino IDE doesn't have an "install from GitHub" feature? 

I'm curious why this library isn't included in the Arduino IDE library manager, given that it seems to be quite popular.

PaulStoffregen, 2018-12-30:
> The library manager (at least for Arduino IDE v 1.9.0-beta) does not include a "Time" or "TimeLib" library

Yes, it does indeed have Time in the library manager.

[image]

Github issues are the **WRONG** place to ask for technical support.  Ask on the Arduino Forum.

https://forum.arduino.cc/index.php

jareklupinski, 2019-07-20:
Had this issue today, was confused because I didn't recognize the author "Michael Margolis", which made me believe that the Time library in the Arduino LIbrary List was not from you Paul. Is it possible to change the "By" name in the Library List to yours?

OscarVanL, 2020-04-03:
Same confusion here. An installation header on the repo readme would be nice.

PaulStoffregen, 2020-04-04:
Also added this to the readme.  https://github.com/PaulStoffregen/Time/commit/ebda9bdf134913ee8384e93dd2281994af14d796

Hopefully these 2 updates to the readme fully resolve this issue?

lmatter, 2021-01-02:
In case someone else arrives here, searching the library manager for the word "time" lists just about every library.  Searching for the word "timekeeping" will show the correct library.

filmonghebre, 2021-03-06:
> In case someone else arrives here, searching the library manager for the word "time" lists just about every library. Searching for the word "timekeeping" will show the correct library.
This helps me to add my TimeLib.h Thank you Immatter

SurenBono, 2022-11-10:
Got it working with copy & pasting the src into the project folder..but I still can't figure out how to blink the separators (:) every 500millis..esp-now stratum2 gps broadcast project on esp8266...

commit 416d009d (2020-04-04): Update readme with include header, fixes #101