PaulStoffregen/Time #107: Error: 'time_t' has not been declared  ->After upgrade IDE

lcorreaesbr, 2018-10-26:
Hi, I´m having problems to compile my program after upgrade from IDE 1.8.5 to 1.8.6.

My program use <TimeLib.h> and <DS1307RTC.h> 

It was working ok, but after upgrade now I get:

error: 'time_t' has not been declared

void store_result(int codeexam, int result, time_t timestart, time_t timeend)

Thanks in advance!

lcorreaesbr, 2018-10-26:
Ok, no problems,  I just post here because I belived that this was an issue, since It was working at first time.

With forum should I ask for your help?

Thanks in advance.

PaulStoffregen, 2018-10-27:
Try Arduino's forum first.  When you post, show the complete code and the exact error, and give details like which board you're using.  Nobody can see your screen, nobody can guess these important details over the internet.  Post a well composed question on the forum to maximize your chance someone will be able to help.

Only open issues here if you're certain the problem is caused by a defect in the software.

closed 2018-10-27 by PaulStoffregen