PaulStoffregen/Time #110: Errors when used in ESP8266 Arduino Core master branch from GitHub

mikekgr, 2019-01-18:
Dear Sirs,
recently I sync my ESP8266 Arduino Core from theirs GitHub master branch.
I discovered that, in all of my sketches that I use your library ( #include <TimeLib.h> ) I got errors at the compilation, even with the simpler sketch ex. : 
`
/*
  Blynk can provide your device with time data, like an RTC.
  Please note that the accuracy of this method is up to several seconds.

  App project setup:
    RTC widget (no pin required)
    Value Display widget on V1
    Value Display widget on V2

  WARNING :
  For this example you'll need Time keeping library:
    https://github.com/PaulStoffregen/Time

  This code is based on an example from the Time library:
    https://github.com/PaulStoffregen/Time/blob/master/examples/TimeSerial/TimeSerial.ino
 *************************************************************/

#include <ESP8266WiFi.h>
#include <BlynkSimpleEsp8266.h>
#include <TimeLib.h>
#include <WidgetRTC.h>

// You should get Auth Token in the Blynk App.
// Go to the Project Settings (nut icon).
char auth[] = "YourAuthToken";

// Your WiFi credentials.
// Set password to "" for open networks.
char ssid[] = "YourNetworkName";
char pass[] = "YourPassword";

BlynkTimer timer;

WidgetRTC rtc;

// Digital clock display of the time
void clockDisplay()
{
  // You can call hour(), minute(), ... at any time
  // Please see Time library examples for details

  String currentTime = String(hour()) + ":" + minute() + ":" + second();
  String currentDate = String(day()) + " " + month() + " " + year();
  Serial.print("Current time: ");
  Serial.print(currentTime);
  Serial.print(" ");
  Serial.print(currentDate);
  Serial.println();

  // Send time to the App
  Blynk.virtualWrite(V1, currentTime);
  // Send date to the App
  Blynk.virtualWrite(V2, currentDate);
}

void setup()
{
  // Debug console
  Serial.begin(9600);

  Blynk.begin(auth, ssid, pass);
  // You can also specify server:
  //Blynk.begin(auth, ssid, pass, "blynk-cloud.com", 80);
  //Blynk.begin(auth, ssid, pass, IPAddress(192,168,1,100), 8080);

  // Begin synchronizing time
  rtc.begin();

  // Other Time library functions can be used, like:
  //   timeStatus(), setSyncInterval(interval)...
  // Read more: http://www.pjrc.com/teensy/td_libs_Time.html

  // Display digital clock every 10 seconds
  timer.setInterval(10000L, clockDisplay);
}

void loop()
{
  Blynk.run();
  timer.run();
}
`

The errors that I got are:
In file included from E:\ARDUINO_ESP_DEV\arduino-IDE\hardware\esp8266com\esp8266/tools/sdk/libc/xtensa-lx106-elf/include/string.h:163:0,

                 from E:\ARDUINO_ESP_DEV\arduino-IDE\portable\sketchbook\libraries\Time\DateStrings.cpp:22:

E:\ARDUINO_ESP_DEV\arduino-IDE\hardware\esp8266com\esp8266/tools/sdk/libc/xtensa-lx106-elf/include/sys/string.h:32:0: warning: "strcpy_P" redefined [enabled by default]

 #define strcpy_P(dest, src)          strncpy_P((dest), (src), SIZE_IRRELEVANT)

 ^

E:\ARDUINO_ESP_DEV\arduino-IDE\portable\sketchbook\libraries\Time\DateStrings.cpp:20:0: note: this is the location of the previous definition

 #define strcpy_P(dest, src) strcpy((dest), (src))

Please help me, what is so wrong?

Thanks and Best Regards,
Mike Kranidis

PaulStoffregen, 2019-01-21:
Maybe try an older version of ESP8266's core library?  Perhaps they changed something?

Best to ask on ESP's issue tracker.  I don't directly support ESP chips.  I'm depending on people from the ESP community to submit pull requests for this sort of issue.

mikekgr, 2019-01-21:
Dear @PaulStoffregen indeed they have changed many things in the forthcoming 2.5.0 version of ESP 8266 Arduino Core. The only problem is this define line at your file: DateStrings.cpp
#define strcpy_P(dest, src) strcpy((dest), (src))
If I put in comment everything is back to normal.
I asked for suggestion on the ESP8266 Arduino Core Github and they answered me to put the indicated line between if... taking in consideration the ESP8266 platform, so please include this very simple logic to exclude this like when the hardware platform is ESP8266. Other than that, everything are working fine.
Thanks for your consideration.

Best Regards,
Mike Kranidis

PaulStoffregen, 2019-03-14:
Merged #120.  Can anyone else with ESP8266 confirm the fix works?

Bolukan, 2019-03-14:
I confirm. But I know, I am not 'anyone else'

danbates2, 2019-05-31:
https://github.com/PaulStoffregen/Time/pull/132