PaulStoffregen/Time #111: Problem witch 2.5.0-beta2 sdk ==> redefined "#define strcpy_P(...."

micha6554, 2019-01-26:
C:\Users\kvms\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2/tools/sdk/libc/xtensa-lx106-elf/include/sys/string.h:32:0: warning: "strcpy_P" redefined [enabled by default]

 #define strcpy_P(dest, src)          strncpy_P((dest), (src), SIZE_IRRELEVANT)

C:\Users\ms\AppData\Local\Arduino.sketchbook\libraries\Time-master\DateStrings.cpp:20:0: note: this is the location of the previous definition

 #define strcpy_P(dest, src) strcpy((dest), (src))

Thanks

PaulStoffregen, 2019-01-28:
Please confirm whether this solves the problem?
https://github.com/PaulStoffregen/Time/commit/8ac0830c1e6f3cd627582f0085aa171161f6bbc2

mikekgr, 2019-01-28:
Dear Sirs,
because I faced on this problem on timelib library, then the bellow solved the problem, just to put in comments the bellow define at 20 line of DateStrings.cpp file that is belong to Time keeping library https://github.com/PaulStoffregen/Time
/// NO NEEDED THIS FOR NEW ESP8266 /// #define strcpy_P(dest, src) strcpy((dest), (src))

Thanks

mikekgr, 2019-02-04:
the bellow seems to work ok for ESP8266 target platform:
`
#ifndef ESP8266
#define strcpy_P(dest, src) strcpy((dest), (src))
#endif
`

danbates2, 2019-05-31:
I'm still having this issue on 2.4.2.
I can't use 2.5.x as there are breaking changes.

Any way to make this work?

Arduino IDE 1.8.9

danbates2, 2019-05-31:
https://github.com/PaulStoffregen/Time/pull/132

mitupaca, 2019-06-18:
> I'm still having this issue on 2.4.2.
> I can't use 2.5.x as there are breaking changes.
> 
> Any way to make this work?
> 
> Arduino IDE 1.8.9

Try this!
In the DateStrings.cpp file, change line 20.

`#ifndef ESP8266`

replaced by

`#ifdef ESP8266`