PaulStoffregen/Time #119: Serial.available not working with TimeTeensy3.ino

AntoineTurmel, 2019-02-28:
### Description

When using the TimeTeensy3.ino script, the code is not going into the Serial.available if block.
But I can set the time manually using an unix time

### Hardware & Software

Board Teensy 3.2
Arduino IDE version 1.8.8
Teensyduino version (if using Teensy) 1.45
Operating system & version Both Ubuntu and Windows 10
Any other software or hardware? No

### Arduino Sketch
TimeTeensy3.ino from this repository

### Errors or Incorrect Output
No error, it's just not going into the Serial.available if block.

PaulStoffregen, 2019-02-28:
I tested just now.  Can't reproduce the problem.

I added "pinMode(13, OUTPUT);" in setup and "digitalWrite(13, HIGH); // turn on LED if we get here" just after Serial.available().

When I open the Arduino Serial Monitor, I see it printing the time.  When I type "hello" and click the send button, the LED turns on.

[image]

Here's the complete code I ran.  I used Ubuntu 18.04, 64 bit.

```
/*
 * TimeRTC.pde
 * example code illustrating Time library with Real Time Clock.
 * 
 */

#include <TimeLib.h>

void setup()  {
  pinMode(13, OUTPUT);
  // set the Time library to use Teensy 3.0's RTC to keep time
  setSyncProvider(getTeensy3Time);

  Serial.begin(115200);
  while (!Serial);  // Wait for Arduino Serial Monitor to open
  delay(100);
  if (timeStatus()!= timeSet) {
    Serial.println("Unable to sync with the RTC");
  } else {
    Serial.println("RTC has set the system time");
  }
}

void loop() {
  if (Serial.available()) {
    digitalWrite(13, HIGH); // turn on LED if we get here
    time_t t = processSyncMessage();
    if (t != 0) {
      Teensy3Clock.set(t); // set the RTC
      setTime(t);
    }
  }
  digitalClockDisplay();  
  delay(1000);
}

void digitalClockDisplay() {
  // digital clock display of the time
  Serial.print(hour());
  printDigits(minute());
  printDigits(second());
  Serial.print(" ");
  Serial.print(day());
  Serial.print(" ");
  Serial.print(month());
  Serial.print(" ");
  Serial.print(year()); 
  Serial.println(); 
}

time_t getTeensy3Time()
{
  return Teensy3Clock.get();
}

/*  code to process time sync messages from the serial port   */
#define TIME_HEADER  "T"   // Header tag for serial time sync message

unsigned long processSyncMessage() {
  unsigned long pctime = 0L;
  const unsigned long DEFAULT_TIME = 1357041600; // Jan 1 2013 

  if(Serial.find(TIME_HEADER)) {
     pctime = Serial.parseInt();
     return pctime;
     if( pctime < DEFAULT_TIME) { // check the value is a valid time (greater than Jan 1 2013)
       pctime = 0L; // return 0 to indicate that the time is not valid
     }
  }
  return pctime;
}

void printDigits(int digits){
  // utility function for digital clock display: prints preceding colon and leading 0
  Serial.print(":");
  if(digits < 10)
    Serial.print('0');
  Serial.print(digits);
}

```

PaulStoffregen, 2019-02-28:
Maybe you could explain in more detail exactly *how* you are testing?  Or try as I did just now, with the LED and sending any text from Arduino Serial Monitor with the "send" button.

AntoineTurmel, 2019-02-28:
I'm testing this on behalf @airel46 instead of your digital write to light a LED I just put a ```
Serial.println("test");``` to see if it's going inside the if block
The setup routine is working because we have ""RTC has set the system time"
I suspect ```Serial.available()``` is not working. Could it be an issue with the board serial ? To be sure we tried on both Windows 10 and Ubuntu but we have the same issue.

PaulStoffregen, 2019-02-28:
Serial.available() does work, as my test with digitalWrite shows.

If you want for me to run a different test, please provide the complete code I should copy into Arduino and run here on a Teensy 3.2.  If sending data is done some other way than "send" in the Arduino Serial Monitor, please be specific and precise in your description of *exactly* how to do the test.

AntoineTurmel, 2019-03-08:
Seems like it's now working for @AIREL46 with ```if (Serial.available() > 0)```
Thanks for your help @PaulStoffregen

closed 2019-03-08 by AntoineTurmel