PaulStoffregen/Time #123: playground URL needs to be capitalised

thatpixguy, 2019-03-22:
The playground.arduino.cc URL's appear to be case sensitive now, so the URL in the description of this git repo returns a broken version of the wiki front page. With a capitalised version of the URL https://playground.arduino.cc/Code/Time/ (captial C and capital T) the correct content is returned.

Also, it could be helpful to mention somewhere that this library appears in the Arduino IDE Library Manager as "Time" by "Michael Margolis", making it easy to miss if scanning for Paul Stoffregen. Also the "More info" link in that entry links to the broken (lower case) URL I mentioned above, so the URLs in the library.properties file might need updating too.

In a perfect world the Arduino site would handle unambiguous lower case URL's and the Library Manager search would be better ;)

ArminJo, 2019-05-10:
Hello Paul,
please update the URL, sine I also stumbled upon this and it took me 1 hour to find the right location.
Or use your influence and convince the Arduino guys to switch back to case insensitve URLs, since most of the other libraries have the same issue.
Thanks
Armin

PaulStoffregen, 2019-05-10:
@per1234 - Is there a special process for updating the playground pages?

per1234, 2019-05-10:
>Is there a special process for updating the playground pages?

The Arduino Playground was recently changed from a public wiki to read-only. So the process for updating the Playground pages is now the same as for updating any other content on the arduino.cc website: submit an issue report to https://github.com/arduino/Arduino/issues and then wait for someone from Arduino with the write access to make the requested change.

closed 2019-05-11 by PaulStoffregen