PaulStoffregen/Time #125: DateStrings.cpp Error in function parameter and declaration.

abhijitraijada, 2019-03-31:
**while compiling:**
C:\Users\Abhijit\Documents\Arduino\libraries\Time-master\DateStrings.cpp: In function 'char* monthStr(uint8_t)':

C:\Users\Abhijit\Documents\Arduino\libraries\Time-master\DateStrings.cpp:76:66: error: 'strcpy_P' was not declared in this scope
 strcpy_P(buffer, (PGM_P)pgm_read_word(&(monthNames_P[month])));

C:\Users\Abhijit\Documents\Arduino\libraries\Time-master\DateStrings.cpp: In function 'char* dayStr(uint8_t)':

C:\Users\Abhijit\Documents\Arduino\libraries\Time-master\DateStrings.cpp:90:61: error: 'strcpy_P' was not declared in this scope
strcpy_P(buffer, (PGM_P)pgm_read_word(&(dayNames_P[day])));

exit status 1
Error compiling for board NodeMCU 0.9 (ESP-12 Module).

Arduino Code:
//Checking for library compilation
#include <Time.h>
#include <TimeLib.h>

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}

Arduino IDE version: 1.8.9

Header files used
Time.h
TimeLib.h

danbates2, 2019-05-31:
Same problem, trying to compile this example: https://github.com/gmag11/NtpClient/tree/master/examples/NTPClientESP8266
ESP8266 used here, platform version 2.4.2.
v2.5.x has major issues and I'm not using it.
Arduino IDE 1.8.9

danbates2, 2019-05-31:
https://github.com/PaulStoffregen/Time/pull/132