PaulStoffregen/Time #134: calculate seconds with two dates

selfmade01, 2019-06-18:
Hello

i use 
timelib.h and mealarrms.h
i have t calculate seconds between two dates...
1) i use the date and time from ntp and my hour i have in hour();
2) i would like calculate a second time and need the seonds

sample:
now is 18.june.2019 22:22.00 
this time i have and the unixtimestamp too

i would like give now a new date like 01.may.2019 20:00:00 ,,, and there i need the unixtimestamp ,,,
if this possible?

kind regards
Achim

KCRAJU12, 2019-06-18:
Hi my Arduino uno skips the data collection why?

PaulStoffregen, 2019-06-26:
This is not the place to ask for help.  Ask on the forums.

If the problem is confirmed to be a bug in this library, then open an issue here with a detailed bug report.  At a bare minimum, a bug report must provide the Arduino code to reproduce the bug.

closed 2019-06-26 by PaulStoffregen