PaulStoffregen/Time #146: Not compiling with Arduino Nano BLE 33

GitLorenzo, 2020-03-28:
### Description

Hello,
The time library does not work on the new Arduino Nano BLE 33 board.

### Steps To Reproduce Problem
When I just upload an empty code with only:
#include <TimeLib.h> I get an errors (see below)

### Hardware & Software

**Board:** Arduino Nano 33 BLE
**Arduino IDE version:** 1.8.12
**Version info:** Time library v 1.6.8
**Operating system & version:** Mac OS X

### Arduino Sketch

```cpp

#include <TimeLib.h>

void setup() {
}

void loop() {
}
```

### Errors or Incorrect Output

Arduino: 1.8.12 (Mac OS X), Board: "Arduino Nano 33 BLE"

In file included from /Users/user/Library/Arduino15/packages/arduino/hardware/mbed/1.1.3/cores/arduino/api/String.h:33:0,
                 from /Users/user/Library/Arduino15/packages/arduino/hardware/mbed/1.1.3/cores/arduino/api/Print.h:24,
                 from /Users/user/Library/Arduino15/packages/arduino/hardware/mbed/1.1.3/cores/arduino/api/Stream.h:25,
                 from /Users/user/Library/Arduino15/packages/arduino/hardware/mbed/1.1.3/cores/arduino/api/Client.h:22,
                 from /Users/user/Library/Arduino15/packages/arduino/hardware/mbed/1.1.3/cores/arduino/api/ArduinoAPI.h:29,
                 from /Users/user/Library/Arduino15/packages/arduino/hardware/mbed/1.1.3/cores/arduino/Arduino.h:42,
                 from /Users/user/Documents/Arduino/libraries/Time-master/DateStrings.cpp:12:
/Users/user/Documents/Arduino/libraries/Time-master/DateStrings.cpp: In function 'char* monthStr(uint8_t)':
/Users/user/Library/Arduino15/packages/arduino/hardware/mbed/1.1.3/cores/arduino/api/deprecated-avr-comp/avr/pgmspace.h:106:49: error: 'const void*' is not a pointer-to-object type
 #define pgm_read_ptr(addr) (*(const void *)(addr))
                                                 ^
/Users/user/Library/Arduino15/packages/arduino/hardware/mbed/1.1.3/cores/arduino/api/deprecated-avr-comp/avr/pgmspace.h:63:45: note: in definition of macro 'strcpy_P'
 #define strcpy_P(dest, src) strcpy((dest), (src))
                                             ^~~
/Users/user/Documents/Arduino/libraries/Time-master/DateStrings.cpp:72:29: note: in expansion of macro 'pgm_read_ptr'
     strcpy_P(buffer, (PGM_P)pgm_read_ptr(&(monthNames_P[month])));
                             ^~~~~~~~~~~~
/Users/user/Documents/Arduino/libraries/Time-master/DateStrings.cpp: In function 'char* dayStr(uint8_t)':
/Users/user/Library/Arduino15/packages/arduino/hardware/mbed/1.1.3/cores/arduino/api/deprecated-avr-comp/avr/pgmspace.h:106:49: error: 'const void*' is not a pointer-to-object type
 #define pgm_read_ptr(addr) (*(const void *)(addr))
                                                 ^
/Users/user/Library/Arduino15/packages/arduino/hardware/mbed/1.1.3/cores/arduino/api/deprecated-avr-comp/avr/pgmspace.h:63:45: note: in definition of macro 'strcpy_P'
 #define strcpy_P(dest, src) strcpy((dest), (src))
                                             ^~~
/Users/user/Documents/Arduino/libraries/Time-master/DateStrings.cpp:86:28: note: in expansion of macro 'pgm_read_ptr'
    strcpy_P(buffer, (PGM_P)pgm_read_ptr(&(dayNames_P[day])));
                            ^~~~~~~~~~~~
exit status 1
Error compiling for board Arduino Nano 33 BLE.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Blubberblub, 2020-09-03:
I have the same problem and the same error. Any idea what a solution could be?

manchoz, 2020-10-09:
The upcoming version of the Arduino Cor for MbedOS-based boards will fix the issue (see. https://github.com/arduino/ArduinoCore-API/commit/cd4ba4ca9ebc3421c65d3b19abe60820b8c4cf25#diff-7cf2fd59cb024f927f2c406eea1d7cc7).