PaulStoffregen/Time #149: Time lib incompatible with nRF52's adafruitadafruitnrf52 framework

orrmany, 2020-05-10:
### Description

DateStrings.cpp fails to compile

### Steps To Reproduce Problem

Try to compile for for any nrF52 board

### Hardware & Software

Board = any nRF52 board
Shields / modules used: not relevant, since this is compile time error
Arduino IDE version: n/a (using PlatformIO)
Teensyduino version (if using Teensy): n/a
Version info & package name (from Tools > Boards > Board Manager) not relevant, this is pure C++ problem
Operating system & version: not relevant
Any other software or hardware? not relavant

### Arduino Sketch

```cpp
// Change the code below by your sketch (please try to give the smallest code which demonstrates the problem)
#include <TimeLib.h>    // https://github.com/PaulStoffregen/Time

void setup() {
}

void loop() {
}
```

### Errors or Incorrect Output
> Executing task in folder SD-card-PoC: C:\Users\egbozie\.platformio\penv\Scripts\platformio.exe run <

Processing adafruit_feather_nrf52840_s340 (platform: https://github.com/orrmany/platform-nordicnrf52.git#develop-s340; board: adafruit_feather_nrf52840_s340; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/adafruit_feather_nrf52840_s340.html
PLATFORM: Nordic nRF52 4.2.0 #3db8a48 > Adafruit Feather nRF52840 Express s340
HARDWARE: NRF52840 64MHz, 243KB RAM, 796KB Flash
DEBUG: Current (jlink) External (jlink, stlink)
PACKAGES:
 - framework-arduinoadafruitnrf52 5aaf2fb
 - tool-sreccat 1.164.0 (1.64)
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 31 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Time> 1.6
Building in release mode
Compiling .pio\build\adafruit_feather_nrf52840_s340\src\main.cpp.o
Compiling .pio\build\adafruit_feather_nrf52840_s340\libbe2\Time_ID44\DateStrings.cpp.o
Compiling .pio\build\adafruit_feather_nrf52840_s340\libbe2\Time_ID44\Time.cpp.o
Compiling .pio\build\adafruit_feather_nrf52840_s340\FrameworkArduinoVariant\variant.cpp.o
Compiling .pio\build\adafruit_feather_nrf52840_s340\FrameworkArduino\HardwarePWM.cpp.o
Compiling .pio\build\adafruit_feather_nrf52840_s340\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\adafruit_feather_nrf52840_s340\FrameworkArduino\Print.cpp.o
Compiling .pio\build\adafruit_feather_nrf52840_s340\FrameworkArduino\RingBuffer.cpp.o
Compiling .pio\build\adafruit_feather_nrf52840_s340\FrameworkArduino\Stream.cpp.o
Archiving .pio\build\adafruit_feather_nrf52840_s340\libFrameworkArduinoVariant.a
In file included from C:\Users\egbozie\.platformio\packages\framework-arduinoadafruitnrf52@src-86afce17ddb3b6c25ca6e2f29943ab03\cores\nRF5/Arduino.h:17:0,
                 from C:\Users\egbozie\.platformio\lib\Time_ID44\DateStrings.cpp:12:
C:\Users\egbozie\.platformio\lib\Time_ID44\DateStrings.cpp: In function 'char* monthStr(uint8_t)':
C:\Users\egbozie\.platformio\packages\framework-arduinoadafruitnrf52@src-86afce17ddb3b6c25ca6e2f29943ab03\cores\nRF5/avr/pgmspace.h:105:49: error: 'const void*' is not a pointer-to-object
type
 #define pgm_read_ptr(addr) (*(const void *)(addr))
                                                 ^
C:\Users\egbozie\.platformio\packages\framework-arduinoadafruitnrf52@src-86afce17ddb3b6c25ca6e2f29943ab03\cores\nRF5/avr/pgmspace.h:63:45: note: in definition of macro 'strcpy_P'
 #define strcpy_P(dest, src) strcpy((dest), (src))
                                             ^~~
C:\Users\egbozie\.platformio\lib\Time_ID44\DateStrings.cpp:72:29: note: in expansion of macro 'pgm_read_ptr'
     strcpy_P(buffer, (PGM_P)pgm_read_ptr(&(monthNames_P[month])));
                             ^~~~~~~~~~~~
C:\Users\egbozie\.platformio\lib\Time_ID44\DateStrings.cpp: In function 'char* dayStr(uint8_t)':
C:\Users\egbozie\.platformio\packages\framework-arduinoadafruitnrf52@src-86afce17ddb3b6c25ca6e2f29943ab03\cores\nRF5/avr/pgmspace.h:105:49: error: 'const void*' is not a pointer-to-object
type
 #define pgm_read_ptr(addr) (*(const void *)(addr))
                                                 ^
C:\Users\egbozie\.platformio\packages\framework-arduinoadafruitnrf52@src-86afce17ddb3b6c25ca6e2f29943ab03\cores\nRF5/avr/pgmspace.h:63:45: note: in definition of macro 'strcpy_P'
 #define strcpy_P(dest, src) strcpy((dest), (src))
                                             ^~~
C:\Users\egbozie\.platformio\lib\Time_ID44\DateStrings.cpp:86:28: note: in expansion of macro 'pgm_read_ptr'
    strcpy_P(buffer, (PGM_P)pgm_read_ptr(&(dayNames_P[day])));
                            ^~~~~~~~~~~~
*** [.pio\build\adafruit_feather_nrf52840_s340\libbe2\Time_ID44\DateStrings.cpp.o] Error 1

orrmany, 2020-05-10:
Sorry, it rather seems to be a framework problem

closed 2020-05-10 by orrmany

schmidie, 2020-10-27:
@orrmany, did you find a way to get the TimeLib working on nRf52 ?

orrmany, 2020-10-27:
Yes, by fixing the framework. See https://github.com/adafruit/Adafruit_nRF52_Arduino/issues/502

orrmany, 2020-10-27:
See also: https://github.com/sandeepmistry/arduino-nRF5/pull/418

schmidie, 2020-10-27:
:+1: cool, thanks