PaulStoffregen/Time #151: Timelib breaks regular time functions on windows

FrankyBoy, 2020-06-27:
I guess this thread has the explanation: https://community.platformio.org/t/build-error-if-time-library-is-installed/450

> I suppose you are a Windows user? Windows system is not a case-sensitive so when your source file contains the line #include <time.h> PlatformIO decides to use library 44 because it has the similar file Time.h. As a workaround, you can just rename file Time.h to _Time.h in library directory c:\Users\your_user\.platformio\lib\Time_ID44 and your project should compile without errors.

Time.h is deprecated anyhow (only contains a compile time error message and nothing more), so please just get rid of it finally.

sijanec, 2021-05-22:
Another solution would be to get rid of your deprecated operating system that is causing problems. (;

jamesmyatt, 2021-07-21:
This can be closed: https://github.com/PaulStoffregen/Time/commit/b0cfc3c941a795672d44db94c3d2d341887bd42a