PaulStoffregen/Time #159: Time.h and time.h issue on windows

mars000, 2021-06-20:
because esp32 already has a built in time.h it gets confused with Time.h as it does not recognize cases.
can the developer pls fix this issues once and for all so it works correctly on windows os ?

blackketter, 2021-06-20:
Use `#include "TimeLib.h"` instead.

PaulStoffregen, 2021-06-21:
Sadly, this is confirmation we *still* have a lot of code in the wild which includes Time.h.  :-(

PaulStoffregen, 2021-06-21:
I am planning to delete Time.h.  That's why Time.h gives a warning to use TimeLib.h.

But this issue is also confirmation that obviously a lot of Arduino users have the default setting where Arduino doesn't show the compiler's warning messages, only errors.

Maybe I'll change the warning to an error and publish a new version.  Hopefully **that** will finally get the remaining Time.h includes changed?

PaulStoffregen, 2021-06-21:
Ok... done, 1.6.1 release!

closed 2021-06-21 by PaulStoffregen