PaulStoffregen/Time #163: TimeLib conflicts with NeoGPS library

BruceScoville, 2021-07-20:
### Description

Recently added a TinyRTC board to my growing project.  I downloaded TimeLib.h and DS1307RTC.
The examples of SET TIME and READ TIME work as described.
When I added just the 2 following statements:
#include <TimeLib.h>
#include <DS1307RTC.h>
to my code I received a compile error:

In file included from C:\Users\admin\Documents\Arduino\libraries\DS1307RTC/DS1307RTC.h:9:0,
                 from C:\Users\admin\Documents\Arduino\B1_FieldTest_2020_12_28\B1_FieldTest_2020_12_28.ino:33:
C:\Users\admin\Documents\Arduino\libraries\Time/TimeLib.h:70:32: error: expected ')' before '(' token
 #define DAYS_PER_WEEK ((time_t)(7UL))
                                ^
C:\Users\admin\Documents\Arduino\libraries\NeoGPS\src/NeoTime.h:44:16: note: in expansion of macro 'DAYS_PER_WEEK'
 const uint8_t  DAYS_PER_WEEK      = 7;
                ^~~~~~~~~~~~~

It looks to me like there is a conflict between TimeLib and the NeoGPS library I have already installed in my project.

### Steps To Reproduce Problem

When I comment-out the TimeLib and DS1307RTC includes, the compile error stops. 

### Hardware & Software

Board: Arduino Mega2560
Shields / modules used: GT-U7 gps; TinyRTC, 2004 lcd display (I2c)
Arduino IDE version: 1.8.15
Version info & package name (from Tools > Boards > Board Manager): Arduino Mega or Mega 2560
Operating system & version Windows 10 
Any other software or hardware?

### Arduino Sketch
(by UnCommenting TimeLib.h and DS1307RTC this will have a complie error
========================================
#include <EEPROMex.h>
#include <EEPROMVar.h>
#include <Arduino.h>
#include <menu.h>
#include <Wire.h>
#include <LiquidCrystal_I2C.h>
//#include <TimeLib.h>
//#include <DS1307RTC.h>
#include <NMEAGPS.h>
#include <GPSport.h>
#include <JC_Button.h>
#include <SPI.h>
#include <SD.h>

void setup(){
}
void loop(){
}

### Errors or Incorrect Output

In file included from C:\Users\admin\Documents\Arduino\libraries\DS1307RTC/DS1307RTC.h:9:0,
                 from C:\Users\admin\Documents\Arduino\B1_FieldTest_2020_12_28\B1_FieldTest_2020_12_28.ino:33:
C:\Users\admin\Documents\Arduino\libraries\Time/TimeLib.h:70:32: error: expected ')' before '(' token
 #define DAYS_PER_WEEK ((time_t)(7UL))
                                ^
C:\Users\admin\Documents\Arduino\libraries\NeoGPS\src/NeoTime.h:44:16: note: in expansion of macro 'DAYS_PER_WEEK'
 const uint8_t  DAYS_PER_WEEK      = 7;
                ^~~~~~~~~~~~~

richteel, 2021-09-14:
Yes, I can confirm that the same issue occurs with Teensy 2.0 but commenting out line 70 of TimeLib.h seems to work. Not certain what issues that may cause though.

NOTE: I do have TimLib and NeoGPS being used in my example TimeGPS_Neo at https://github.com/richteel/Time/ and it works fine. I believe the issue is when calling certain code from NeoGPS. I plan to investigate further and plan to reply when I find the cause.

I suspect that SlashDevin and Paul may need to discuss to deconflict the two but we will see.

richteel, 2021-09-15:
I will be creating a pull request in the NeoGPS project. The following is the comment that I added to the NeoGPS issue. 

tscofield made a similar pull request however I wanted to make it so the same name may be used by both libraries. This only has the disadvantage if the days of the week ever change from 7.

Below is the commit statement from my change in NeoGPS.

TimeLib defines DAYS_PER_WEEK as type time_t with a value of 7. TimeLib's time_t is defined as unsigned long with a value of 7. This is compatable with NeoGPS declaration of DAYS_PER_WEEK with type uint8_t if cast to uint8_t. I modified the code to check if DAYS_PER_WEEK is defined and cast DAYS_PER_WEEK as uint8_t when used.

This is an alternate way to resolve the conflict between NeoGPS and TimeLib. User, tscofield, proposed a similar fix by renaming DAYS_PER_WEEK but I believe this may be a slightly better resolution to the issue.

See issues

- [https://github.com/SlashDevin/NeoGPS/pull/115](https://github.com/SlashDevin/NeoGPS/pull/115 "NeoGPS Issue 115")
- [https://github.com/PaulStoffregen/Time/issues/163](https://github.com/PaulStoffregen/Time/issues/163 "TimeLib Issue 163")

BTW: I looked into applying a similar fix to TimeLib but the issue only seemed to be resolved when applied to NeoGPS.