PaulStoffregen/Time #180: Epoch and Time offset

AdvdLaar, 2023-05-06:
Please use this form only to report code defects or bugs.

For any question, even questions directly pertaining to this code, post your question on the forums related to the board you are using.

Arduino: forum.arduino.cc
Teensy: forum.pjrc.com
ESP8266: www.esp8266.com
ESP32: www.esp32.com
Adafruit Feather/Metro/Trinket: forums.adafruit.com
Particle Photon: community.particle.io

If you are experiencing trouble but not certain of the cause, or need help using this code, ask on the appropriate forum.  This is not the place to ask for support or help, even directly related to this code.  Only use this form you are certain you have discovered a defect in this code!

Please verify the problem occurs when using the very latest version, using the newest version of Arduino and any other related software.

----------------------------- Remove above -----------------------------

### Description

Describe your problem.
I am using the time library and I am happy with it. There is just 1 thing. I am using your library and wanted to make corrections for the DST or no DST. I set the adjustment with adjustTime. It adds the value to sysTime. I want the adjustTime to be the same after a NTP time callback routine(setSyncProvider routine) . Can you make a global variable TimeOffset that will replace or will be set by adjustTime. I like it to be a global variable, so I can read back the offset elsewhere in the program. The time I will provide the Timelib will be Epoch time. The offset will be used by the return value of now(). return (time_t)(sysTime + TimeOffset);
This way the  the sysTime is epoch and every second that passes adds 1 count to the sysTime EpochTime. By using an offset the time will be correct all the time. if it's between 2 and 3 AM during the last day of Octobre it's not clear if the time if during or after DST, but the epoch time never lies. Has no time correction. So please use a EpochTime sysTime with an offset. I am living in Europe and I think it has other rules as other places in the world. The Epoch DST routine must be done by the end user of this library (in the setSyncProvider callback which gets Epoch time). if DST changes I want to set the other offset with adjustTime or a global variable. sysTime will stay the same, but now() will change it value after the offset will change.
I wrote some DST routines that you can also use. works fine now, but I can't use adjustTime. Epoch time keeps returning ( setTime(TimeData.Epoch)); My TimeData struct contains 4 members (Epoch, DST, Offset and Actual). I can't use this to set time, because my setSyncProvider returns Epoch time and the previous adjustTime offset is disregarded. I also want to be able to get the sysTime Epoch time from the timelib. I am saving all my sensor data in my project onto an SD card with Epoch time codes.
### Steps To Reproduce Problem
Use adjustTime and don't update the time (with setSyncProvider). adjustTime will probably work. If you use a setSyncProvider routine than it doesn't work anymore. the offset is forgotten. problems and solution in text above.
Please give detailed instructions needed for anyone to attempt to reproduce the problem.
problems and solution in text above.

### Hardware & Software
I am using a nodeMCU at the moment.
Board
Shields / modules used WH24P (weather station)
Arduino IDE version 2.1.0
Teensyduino version (if using Teensy)
Version info & package name (from Tools > Boards > Board Manager)
Operating system & version Windows 11
Any other software or hardware?

### Arduino Sketch

```cpp
// Change the code below by your sketch (please try to give the smallest code which demonstrates the problem)
#include <Arduino.h>

// libraries: give links/details so anyone can compile your code for the same result

void setup() {
}

void loop() {
}
```

### Errors or Incorrect Output

If you see any errors or incorrect output, please show it here.  Please use copy & paste to give an exact copy of the message.  Details matter, so please show (not merely describe) the actual message or error exactly as it appears.

AdvdLaar, 2023-05-06:
This is my DST routine with Epoch time from NTP server.

#define nextSunday(_time_) (previousSunday(_time_)+SECS_PER_WEEK)

bool CheckDSTTime(time_t EpochTime)
{
  bool Result = false;
  tmElements_t CurrentTime;
  time_t DSTBegin, DSTEnd;
  breakTime(EpochTime, CurrentTime);
  tmElements_t DSTBeginDate = {0, 0, 0, 0, 24, 3, CurrentTime.Year};
  tmElements_t DSTEndDate = {0, 0, 0, 0, 24, 10, CurrentTime.Year};
  //Calculate Epoch Time when clock changes
  //Starts a 2AM - an hour back from CET-1
  //DST starts at 2.00h in the Netherlands. That's 1.00h Epoch time
  DSTBegin = nextSunday(makeTime(DSTBeginDate)) + 3600UL;
  //Calculate Epoch Time when clock changes
  //Starts a 3AM - an hour from CET-1 - 1 hour of DST difference. That's 1.00h Epoch time
  DSTEnd = nextSunday(makeTime(DSTEndDate)) + 3600UL;
  Result = ((EpochTime >= DSTBegin) && (EpochTime < DSTEnd));
  return (Result);
}

AdvdLaar, 2023-05-06:
I have solved the adjustTime problem and have rewritten now() (to return sysTime + timeOffset) and adjustTime(). I have added nowEpoch() and getAdjustTime(). I didn't rename anything (including adjustTime). The timeOffset will be kept in a long. So now the adjustTime offset will be remembered after a NTP/RTC callback. sysTime is Epoch time. I didn't change anything else. I don't want to use my own library, because I want it to be updateable.

This is Time.cpp

`/*
  time.c - low level time and date functions
  Copyright (c) Michael Margolis 2009-2014

  This library is free software; you can redistribute it and/or
  modify it under the terms of the GNU Lesser General Public
  License as published by the Free Software Foundation; either
  version 2.1 of the License, or (at your option) any later version.

  This library is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
  Lesser General Public License for more details.

  You should have received a copy of the GNU Lesser General Public
  License along with this library; if not, write to the Free Software
  Foundation, Inc., 51 Franklin St, Fifth Floor, Boston, MA  02110-1301  USA
  
  1.0  6  Jan 2010 - initial release
  1.1  12 Feb 2010 - fixed leap year calculation error
  1.2  1  Nov 2010 - fixed setTime bug (thanks to Korman for this)
  1.3  24 Mar 2012 - many edits by Paul Stoffregen: fixed timeStatus() to update
                     status, updated examples for Arduino 1.0, fixed ARM
                     compatibility issues, added TimeArduinoDue and TimeTeensy3
                     examples, add error checking and messages to RTC examples,
                     add examples to DS1307RTC library.
  1.4  5  Sep 2014 - compatibility with Arduino 1.5.7
*/

#if ARDUINO >= 100
#include <Arduino.h> 
#else
#include <WProgram.h> 
#endif

#include "TimeLib.h"

static tmElements_t tm;          // a cache of time elements
static time_t cacheTime;   // the time the cache was updated
static uint32_t syncInterval = 300;  // time sync will be attempted after this many seconds

void refreshCache(time_t t) {
  if (t != cacheTime) {
    breakTime(t, tm); 
    cacheTime = t; 
  }
}

int hour() { // the hour now 
  return hour(now()); 
}

int hour(time_t t) { // the hour for the given time
  refreshCache(t);
  return tm.Hour;  
}

int hourFormat12() { // the hour now in 12 hour format
  return hourFormat12(now()); 
}

int hourFormat12(time_t t) { // the hour for the given time in 12 hour format
  refreshCache(t);
  if( tm.Hour == 0 )
    return 12; // 12 midnight
  else if( tm.Hour  > 12)
[... 407 lines cut ...]
void    setTime(int hr,int min,int sec,int day, int month, int yr);
void    adjustTime(long adjustment);
long    getAdjustTime();

/* date strings */ 
#define dt_MAX_STRING_LEN 9 // length of longest date string (excluding terminating null)
char* monthStr(uint8_t month);
char* dayStr(uint8_t day);
char* monthShortStr(uint8_t month);
char* dayShortStr(uint8_t day);
	
/* time sync functions	*/
timeStatus_t timeStatus(); // indicates if time has been set and recently synchronized
void    setSyncProvider( getExternalTime getTimeFunction); // identify the external time provider
void    setSyncInterval(time_t interval); // set the number of seconds between re-sync

/* low level functions to convert to and from system time                     */
void breakTime(time_t time, tmElements_t &tm);  // break time_t into elements
time_t makeTime(const tmElements_t &tm);  // convert time elements into time_t

} // extern "C++"
#endif // __cplusplus
#endif /* _Time_h */

`

KOEGlike, 2023-12-02:
could you please make a pool request